gdsfactory/gdsfactory #2230: Extend `get_route` to allow different CrossSections for bends and straights

nikosavola, 2023-10-26:
**Is your feature request related to a problem? Please describe.**
Extend `get_route` to allow different CrossSections for bends and straights. This would allow, for example, adding different marker layers through cladding.

tvt173, 2023-10-26:
@nikosavola, to make sure I understand you correctly, you want to be able to use different cross sections for bends vs straight sections within a route?

the way you can do this today is to set the `bend` or `straight` argument to the `get_route` function to a component which effectively overrides the `cross_section` argument priveded to `get_route`.

for example, if you want to have your bends always in strip, make a bend component that is strip but tapers to whatever cross section is passed as an argument. then use that bend component when you call `get_route()`. does that make sense?

nikosavola, 2023-10-26:
Sounds good, I'll try that and close the issue if it works. Thanks!

tvt173, 2023-10-26:
(i do this all the time, so i know it should work. happy to send an example if you need one!)

closed 2023-10-27 by joamatab