gdsfactory/gdsfactory #2252: `gf.config.enable_off_grid_ports` doesn't work for built-in components

wwwwwwzx, 2023-10-31:
**Describe the bug**
It could be those built-in Component functions with `@cell` already saved the default settings into the callable object at the moment initializing the `gdsfactory` module. The changes in `gf.config.CONF` after `import` will not affect the assertions in the cell wrapper.

**To Reproduce**
[This notebook](https://gdsfactory.github.io/gdsfactory/notebooks/rib_strip_autotransition.html) shows the non-manhattan warnings even `enable_off_grid_ports` is called at the beginning.

**Expected behavior**
All warnings are suppressed after `gf.config.enable_off_grid_ports()`

**Suggested fix**
defer the default settings in `@cell` arguments
```python
    ports_off_grid: str = CONF.ports_off_grid,
    ports_not_manhattan: str = CONF.ports_not_manhattan,
```
to
```python
    ports_off_grid: str = "",
    ports_not_manhattan: str = "",
```
and
```python
@functools.wraps(func)
    def wrapper(*args, **kwargs) -> Component:
        nonlocal ports_not_manhattan, ports_off_grid
        ports_off_grid = ports_off_grid or CONF.ports_off_grid
        ports_not_manhattan = ports_not_manhattan or CONF.ports_not_manhattan
```

tvt173, 2023-11-01:
i've also noticed this bug. thanks for tracing the root cause

tvt173, 2023-11-01:
minor notes:

1. The same goes for the `max_name_length` argument
2. I would recommend setting all of these to `None` by default, for consistency and clarity

PR #2254 (2023-11-01): defering evaluation of default args from CONF
Fixes #2252 

Thanks @wwwwwwzx
commit bb954f0e (2023-11-01): defering evaluation of default args from CONF. fixes #2252
closed 2023-11-01 by joamatab