gdsfactory/gdsfactory #2268: kfactory backend

joamatab, 2023-11-03:
kfactory is more actively maintained than gdstk. To fully evaluate [kfactory as backend](https://github.com/gdsfactory/gdsfactory/tree/kfactory). There are a few things that need to be ported to the new syntax

- [ ] gf.routing.get_bundle
- [ ] gf.routing.route_ports_to_side
- [ ] gf.routing.all_angle
- [ ] gf.routing.route_to_fiber_array

@sebastian-goeldi 
@damienbonneau
@tvt173 
@SkandanC

joamatab, 2023-11-06:
Troy, can you review `get_bundle_path_length_match`?
Damien, can you review path_length_match?
Sebastian, can you help with spiral_inner_io and gf.grid?

sebastian-goeldi, 2024-02-10:
@joamatab can we convert this to the https://github.com/orgs/gdsfactory/projects/6/views/1 ?

joamatab, 2024-02-11:
Sounds good to me

sebastian-goeldi, 2024-05-28:
@joamatab can we update this. What's left to do? the path length matching router?

joamatab, 2024-05-28:
I think fixing the tests is the most important,

path length matching would be nice if we can also make it to work

sebastian-goeldi, 2024-05-28:
I think I can implement pathlength matching reasonably into the route_smart logic.

closed 2025-02-19 by sebastian-goeldi