gdsfactory/gdsfactory #2271: Improve repository security

95-martin-orion, 2023-11-03:
**Is your feature request related to a problem? Please describe.**
The Open Source Security Foundation (OSSF) provides a [scorecard tool](https://github.com/ossf/scorecard) for rating the security of a repository based on a wide range of metrics. At the time of writing it gives gdsfactory a score of 5.2/10, citing the following issues (among others):

- Branch protection is not enabled
- 4 of the last 10 PRs were not reviewed
- Dependency versions are not pinned

**Describe the solution you'd like**
Addressing the issues list above would remove a potential barrier to adoption for new users considering gdsfactory.

joamatab, 2023-11-03:
Good points, Thank you

- [x] I just enabled branch protection
- [x] We required 1 approval before mergin
- [x] Every release has a requirements.txt https://github.com/gdsfactory/gdsfactory/releases/tag/v7.8.14. is that good for you?

95-martin-orion, 2023-11-03:
This is great, thanks!

closed 2023-11-04 by joamatab