gdsfactory/gdsfactory #2294: Use of cell decorator on instance methods crashes Python kernel

karl-johnson, 2023-11-08:
I apologize if this has been addressed previously, I searched the documentation and issues but was not able to find anything similar.

**Describe the bug**
Using the cell decorator on an instance method of a class crashes Python. I am doing some MEMS layout and planned on creating a class with instance variables containing all the geometry of the device. Then I can have instance methods to do some analytical modeling of the device, as well as a method to output the layout using GDSfactory. However, when I implemented this, I noticed that using the cell decorator on the instance method to generate the layout causes a crash. I am guessing this is because the cell decorator is failing when it sees the "self" object as an argument to this method (I am not very familiar with the inner workings of GDSfactory but it looks like the arguments are often how @gf.cell determines the cell naming).

**To Reproduce**
The following example code reliably crashes my kernel (gdsfactory 7.3.5, Python 3.11 running in Spyder):
```
import gdsfactory as gf

class fancyRectangle:
    def __init__(self):
        width = 5
        height = 5
    @gf.cell()
    def fancyRectangleLayout(self):
        c = gf.Component()
        myBbox = [[-self.width/2, -self.height/2], [self.width/2, self.height/2]]
        c << gf.components.bbox(bbox=myBbox, layer=(1,0))
    
thisRect = fancyRectangle()
layout = thisRect.fancyRectangleLayout()
layout.show()
```

Here are the first few lines of the crash (it keeps going the same after that):
```
Windows fatal exception: stack overflow

Main thread:
Current thread 0x000040b8 (most recent call first):
  File "[REDACTED]\anaconda3\Lib\site-packages\gdsfactory\serialization.py", line 45 in clean_value_json
  File "[REDACTED]\anaconda3\Lib\site-packages\gdsfactory\serialization.py", line 102 in clean_value_json
  File "[REDACTED]\anaconda3\Lib\site-packages\gdsfactory\serialization.py", line 102 in clean_value_json
  File "[REDACTED]\anaconda3\Lib\site-packages\gdsfactory\serialization.py", line 102 in clean_value_json
...
```

**Expected behavior**
I am not a very experienced Python user, and I do not know if structuring code in this way is the right way of doing things. Is there a better way to group together the code so that I do my device modeling and layout generation from the same set of variables?

Regardless of whether this is a bad way to structure code, maybe it would be beneficial to add some failsafe/error to prevent this crash.

joamatab, 2023-11-08:
why not do this?

```python
import gdsfactory as gf

@gf.cell()
def fancy_rectangle(width=5, height=5):
    c = gf.Component()
    myBbox = [[-self.width / 2, -self.height / 2], [self.width / 2, self.height / 2]]
    c << gf.components.bbox(bbox=myBbox, layer=(1, 0))

c = fancy_rectangle()
c.show()

```

karl-johnson, 2023-11-08:
I understand, and that's what I have done for all the layout I've done with gdsfactory up to this point. This is totally fine for photonic devices, because I can then take the GDS geometry and feed it into a simulation tool like Lumerical. For this specific case, however, I am designing a MEMS device and I am not using simulation tools which take in the whole geometry of the device. Instead I'm doing some basic analytical calculations which of course use the exact same dimensions/parameters that are used to generate the layout. The analog of this in the toy "fancy rectangle" example is this:

```
import gdsfactory as gf

class fancyRectangle:
    def __init__(self):
        width = 5
        height = 5
    @gf.cell()
    def fancyRectangleLayout(self):
        c = gf.Component()
        myBbox = [[-self.width/2, -self.height/2], [self.width/2, self.height/2]]
        c << gf.components.bbox(bbox=myBbox, layer=(1,0))
    
    def computePerimeter(self):
        return 2*(self.width+self.height)
    
    def computeArea(self):
        return self.width*self.height
    
thisRect = fancyRectangle()
layout = thisRect.fancyRectangleLayout()
layout.show()
```

Obviously, for such methods/geometry this doesn't make much sense. But for my device which has 27 different parameters which control the geometry, it would be nice to group the geometry generation and the analytical calculations/estimates so that I don't have a bunch of standalone functions with 30 arguments. I thought making the device a class with layout generation and analytical simulation methods would be the best approach for this. Again, I wouldn't consider myself super familiar with Python/gdsfactory - is there another workflow that is more in line with how GDSfactory is meant to be used?

joamatab, 2023-11-08:
You can take a look at this tutorial
https://gdsfactory.github.io/gdsfactory/notebooks/04_components_hierarchy.html

classes are not recommended in gdsfactory

i also noticed you are missing a return in the function

but anyways with this code you won't be able to cache the class settings and you will get name collisions

karl-johnson, 2023-11-09:
Got it. If classes aren't recommended in gdsfactory I'll close the issue and find a different approach for my particular problem.

closed 2023-11-09 by karl-johnson

tvt173, 2023-11-09:
@karl-johnson , something like this would work, for instance:

```python
import gdsfactory as gf

@gf.cell
def fancy_rectangle(width: float = 5, height: float = 5) -> gf.Component:
    c = gf.Component()
    myBbox = [[-width/2, - height/2], [width/2, height/2]]
    c << gf.components.bbox(bbox=myBbox, layer=(1,0))
    return c

class FancyRectangle:
    def __init__(self):
        self.width = 5
        self.height = 5

    def create_layout(self):
        return fancy_rectangle(width=self.width, height=self.height)
    
    def compute_perimeter(self):
        return 2*(self.width+self.height)
    
    def compute_area(self):
        return self.width*self.height
    
this_rect = FancyRectangle()
this_rect.height = 10
layout = this_rect.create_layout()
layout.show()
```

nikosavola, 2023-11-09:
I kinda like this way, could the @gf.cell also work by using @staticmethod in this case?

> @karl-johnson , something like this would work, for instance:
>
> ```python
> import gdsfactory as gf
>
> @gf.cell
> def fancy_rectangle(width: float = 5, height: float = 5) -> gf.Component:
>     c = [...]

tvt173, 2023-11-09:
@nikosavola , out of curiosity, i tried it, and it does indeed work

```python
import gdsfactory as gf

class FancyRectangle:
    def __init__(self):
        self.width = 5
        self.height = 5

    @gf.cell
    @staticmethod
    def super_fancy_rectangle(width: float = 5, height: float = 5) -> gf.Component:
        c = gf.Component()
        myBbox = [[-width/2, - height/2], [width/2, height/2]]
        c << gf.components.bbox(bbox=myBbox, layer=(1,0))
        return c

    def create_layout(self):
        return FancyRectangle.super_fancy_rectangle(width=self.width, height=self.height)
    
    def compute_perimeter(self):
        return 2*(self.width+self.height)
    
    def compute_area(self):
        return self.width*self.height
    
this_rect = FancyRectangle()
this_rect.height = 10
layout = this_rect.create_layout()
layout.show()
```

_However_, i do kind of like having the simple methods _decoupled_ from the classes which can be used for configuration, etc.

I don't actually use this pattern when using gdsfactory, but I've used it in multiple scientific projects in the past, and I tend to like it. You can kind of get the best of both worlds from object-oriented and functional programming, since the objects make complex configuration, etc. possible, but the critical functions are still stateless, with simple and clear inputs.

wwwwwwzx, 2023-11-09:
Making `FancyRectangle` class serializable by adding the `to_dict` method works

```python
import gdsfactory as gf

class FancyRectangle:
    def __init__(self):
        self.width = 5
        self.height = 5

    def to_dict(self):
        return self.__dict__

    @gf.cell
    def create_layout(self):
        c = gf.Component()
        c.add_polygon([[-self.width/2, - self.height/2], [self.width/2, self.height/2]], layer=(1,0))
        return c
    
    def compute_perimeter(self):
        return 2*(self.width+self.height)
    
    def compute_area(self):
        return self.width*self.height
    
this_rect = FancyRectangle()
this_rect.height = 10
layout = this_rect.create_layout()
layout.show()
```

karl-johnson, 2024-01-10:
@wwwwwwzx The first method you provided has ended up working really well for me on this project, thanks a lot for providing that simple solution. I am commenting again on this thread because I have found another case where putting the geometry generation in a class is incredibly convenient - generating arrays of components which only differ slightly from each other. This comes up frequently when generating arrays of test structures with a variety of dimensions for characterizing fabrication processes.

```python
class TestStructure:
    # parent class for test structure instances
    def to_dict(self):
        return self.__dict__ 
    def to_component():
        raise NotImplementedError("Child class did not implement to_component()!")
    
@gf.cell
def arrayStructure1D(seedComponent: TestStructure, varName, varArray, dispVector):
    # generic function to array a test structure with variations
    # seedComponent: a valid instance of TestStructure
    # varName: the name of the independent variable parameter
    # varArray: an array of values 
    # dispVector: numpy array for x,y displacement between instances
    # convert to np array so we can handle tuple, 
    dispVectorNp = np.array(dispVector)
    c = gf.Component()
    for thisValue in enumerate(varArray):
        setattr(seedComponent, varName, thisValue[1])
        thisInstance = c << seedComponent.to_component()
        thisInstance.move(thisValue[0]*dispVectorNp)
    return c
```