gdsfactory/gdsfactory #2301: Support multiple decorators for a cell initialised with `partial(Component, decorator=[...])`

nikosavola, 2023-11-08:
**Is your feature request related to a problem? Please describe.**
Support multiple decorators instead of only one for a Component initialised like `partial(Component, decorator=...)`. I think this is a natural extensions for the API, although the functionality can already be done using toolz:
```python
from toolz import compose

compose(decorator1, decorator2)
```

**Describe the solution you'd like**
Allow a Sequence of decorators for `decorator`

joamatab, 2023-11-08:
yes, i think compose works but adds more complexity for users

tvt173, 2023-11-09:
i'm in favor of forcing the user to use compose or a good old fashioned pattern like this, rather than adding even more complexity here
```python

def my_multi_decorator(component):
    c = decorator1(component)
    c = decorator2(c)
    return c
```

joamatab, 2023-11-09:
how about adding decorators, which could accept be a list of decorators applied?

tvt173, 2023-11-09:
yeah, something like this, right?

```python
def multi_decorator(*decorators):
    def _multi(component):
        for decorator in decorators:
            component = decorator(component)
        return component
    return _multi

c = my_component(decorator=multi_decorator(decorator1, decorator2))
```

but wouldn't that be basically the same as `compose`?

joamatab, 2023-11-09:
How about ?

```
c = my_component(length=10, decorators=(decorator1, decorator2))
```

tvt173, 2023-11-09:
but why add extra complexity when it's almost as easy and more straightforward to write 

```python
c = my_component(length=10, decorators=multi_decorator(decorator1, decorator2))
```

joamatab, 2023-11-09:
Good point,

we already support

```
    c = straight(width=2, decorator=gf.compose(gf.routing.add_fiber_array, gf.add_tapers))
```

How about we leave it as it is?

nikosavola, 2023-11-09:
I see, compose is available from gdsfactory straight away as well. Maybe it is fine as is.

closed 2023-11-09 by nikosavola