gdsfactory/gdsfactory #2328: Cannot use pip install gdsfactory[cad] --upgrade

Philouchouette, 2023-11-24:
Hi!

I have trouble installing gdsfactory: I succesfully ran the command "conda install -c conda-forge gdsfactory -y", which installed gdsfactory 7.3.5 on my conda environment. However, when I run the command "pip install gdsfactory[cad] --upgrade", I have an error which fails to build wheel for klayout (I use windows 10):

I run: "pip install gdsfactory[cad] --upgrade" on my conda environment. When it is time for the command to build wheel for Klayout:

    Building wheel for klayout (pyproject.toml) ... error
      error: subprocess-exited-with-error
    
      × Building wheel for klayout (pyproject.toml) did not run successfully.
      │ exit code: 1
      ╰─> [41 lines of output]
                ...
                ...
                warning: no files found matching '*.cc' under directory 'src\version'
                warning: no files found matching '*.cc' under directory 'src\plugins\*\db_plugin'
                warning: no files found matching '*.h' under directory 'src\plugins\*\db_plugin'
                warning: no files found matching 'src\py.typed'
                ...
                ...
              error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": 
              https://visualstudio.microsoft.com/visual-cpp-build-tools/
              [end of output]
    
    note: This error originates from a subprocess, and is likely not a problem with pip.
    ERROR: Failed building wheel for klayout
    Failed to build klayout
    ERROR: Could not build wheels for klayout, which is required to install pyproject.toml-based projects

It seems to be an error with communication between python and Cpython. I have then installed Microsoft Visual C++ tools for windows. I have tried many different ways, but the easiest way is to install Visual Studio build tools 2022 (https://visualstudio.microsoft.com/fr/downloads/). Then, on the visual studio installer, select "Workloads → Desktop development with C++", and on the individual components, make sure to select "Windows 10 SDK" and  "C++ x64/x86 build tools", and then install. I have tried many other ways to install c++ tools with Visual Studio, but for all of those ways, once I retry running "pip install gdsfactory[cad] --upgrade", I get the same error, which is that a zlib.h file does not exist when it is time to build the wheel for Klayout:

I run: "pip install gdsfactory[cad] --upgrade", when the build for the wheel for Klayout begins, I get:

  Building wheel for klayout (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for klayout (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [178 lines of output]
            ...
            ...
            src\tl\tl\tlDeflate.cc(30): fatal error C1083: Impossible d'ouvrir le fichier includeÿ: 'zlib.h'ÿ: No such file or directory
            error: command 
            'C:\\ProgramFiles(x86)\\MicrosoftVisualStudio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.38.33130\\bin\\HostX86\\x64\\cl.exe' 
            failed with exit code 2
            [end of output]

   note: This error originates from a subprocess, and is likely not a problem with pip.
   ERROR: Failed building wheel for klayout
   Failed to build klayout
   ERROR: Could not build wheels for klayout, which is required to install pyproject.toml-based projects

I really don't know anything about zlib.h, and I am stuck. Has this problem already happened before? I am running ma annaconda prompt as administrator and I have Klayout installed on my computer.

Thank you!
Philippe

Philouchouette, 2023-11-24:
To be more precise, I just realize that the problem comes when the command "pip install[cad] --upgrade" tries to install the package klayout; I have tried installing it manually and I arrived to the same errors showned earlier.  I use python 3.10 on my conda environment.

tvt173, 2023-11-24:
Which version of klayout? I also had trouble installing klayout 0.28.13 yesterday on windows. But 0.28.12, for instance, works fine. So it may be a klayout bug. Can you try `pip install klayout==0.28.12`? Does that work for you?

Philouchouette, 2023-11-27:
Yes it worked thank you so much!

tvt173, 2023-11-27:
@Philouchouette, FYI, the klayout bug appears to be fixed

closed 2023-11-27 by tvt173