gdsfactory/gdsfactory #2393: get_route_from_steps accepting Components

simbilod, 2023-12-17:
**Is your feature request related to a problem? Please describe.**

get_route_from_steps uses a handy list of dict of step directives `steps = [{"x", 5}, {"dy", 10},...]`. These are currently limited to `x, y, dx, dy`, and are used to wrap route_waypoints (list of coordinates).

This leads to routes containing straights, 90 degree bends, and simple tapers, which are OK most of the time but fall short when more optimized routes are required. The main example I have in mind is when routing U-turns: there, a 180 degree Euler bend would be optimal, as it would save 2 straight --> curved transitions.

**Describe the solution you'd like**

It would be awesome to add a `component` step directive to add more fine grained control over the routes:

```
[{"x", 5}, 
{"dy", 10},
{"component", bend_euler180},
...
]
```

The port centers of the provided 2-port component could be used in the waypoint, but the straight/taper/bend references replaced by this component during generation of the route

joamatab, 2023-12-17:
Sounds like a good idea Simon

tvt173, 2024-02-05:
agreed, this is something that I always wanted to add "sometime in the future" after making `get_route_from_steps()` but never got to... 🙂  

note that something similar _is_ possible with all-angle routes and custom connectors and bends... the difference for connectors is that you pre-allocate the space with the other info in the step and fill in the space with whatever is defined for the connector. so a major restriction is that the connector needs to be straight. on the other hand, when defining your _bend_, you can also have complex logic i.e. to handle 180 degree turns differently than 90 degree turns, differently than 45 degree turns...
https://gdsfactory.github.io/gdsfactory/notebooks/04_routing_non_manhattan.html#bends-and-connectors

sebastian-goeldi, 2026-09-14:
afaik this is done (more or less)

closed 2026-09-14 by sebastian-goeldi