gdsfactory/gdsfactory #2423: Deprecate cell magic methods decorator and info?

joamatab, 2023-12-31:
We do this a lot in gdsfactory

```python
from functools import partial 
import gdsfactory as gf
c = partial(gf.c.mzi, decorator=gf.add_padding)
c = gf.routing.add_fiber_array(c)
c.show()
```

but pyright does not like it :)

[image]

Shall we deprecate this behavior?

sebastian-goeldi, 2023-12-31:
If this is the sole puprose of the partial, why are you using a partial? Just do `add_padding(gf.c.mzi)` or if the decorator takes args `add_padding()(gf.c.mzi)`

joamatab, 2023-12-31:
That was just an example the most common use of the decorator is:

```
wg = gf.components.straight()
wg_with_gc = gf.components.straight(decorator=gf.routing.add_fiber_array)
```

but has the same problems with pyright

i think overall is quite convenient but maybe too hacky/magical?

sebastian-goeldi, 2023-12-31:
Yes, but you can do the same here:

```python
wg = gf.components.straight()

wg_with_gc =gf.routing.add_fiber_array(gf.components.straight)()

```

This might only cause problems if caching is involved. But you are adding this on another hierarchy level anyway if I understand correctly.

For understanding and ease of use I would advocate (and just hide the above "hacky" way) for this:

```python
@gf.routing.add_fiber_array
def wg_with_gc(*args, **kwargs):
    return gf.components.straight(*args, **kwargs)
```

This makes it very easy to explain, as there are tons of good explanations out there as for what a decorator is and how they work.

joamatab, 2024-01-01:
sounds good, 

I deprecated it 

see fixes in https://github.com/gdsfactory/gdsfactory/pull/2421

closed 2024-01-01 by joamatab