gdsfactory/gdsfactory #2451: Change warnings.warn to loguru ones

nikosavola, 2024-01-09:
After https://github.com/gdsfactory/gdsfactory/pull/2442, the logs are doubled with both warning.warn and loguru printing out the same error messgae with default settings.

Additionally, the loguru print doesn't retain information on where the warning was raised and just shows
```
gdsfactory.config:showwarning:281 
```
as the location due to the redirecting.

## Fix

I suggest using exclusively loguru printing for clearer and better manageable logging to for example redirecting to files (`logger.add(file)`)

joamatab, 2024-01-09:
i prefer using warnings as you can treat them as exceptions

feel free to send a PR for this

wwwwwwzx, 2024-01-09:
Use `logger.opt(depth=depth)` at this [line](https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/config.py#L281) is the way to let logger print the useful information [from inside a wrapped function](https://loguru.readthedocs.io/en/stable/api/logger.html#loguru._logger.Logger.opt).

yaugenst, 2024-01-12:
Fixed by #2442

closed 2024-01-12 by yaugenst