gdsfactory/gdsfactory #2462: `gf` CLI commands use dash and underscore inconsistently

nikosavola, 2024-01-10:
**Describe the bug**
This is a very small thing but the gf CLI subcommands use both dash and underscore. Would make sense to use only one to use.

```
─ Commands ─────────────────────────────────────╮
│ from_updk                   Writes a PDK in    │
│                             python from uPDK   │
│                             YAML spec.         │
│ gds-diff                    Show boolean       │
│                             difference between │
│                             two GDS files.     │
│ install-git-diff            Install git diff.  │
│ install-klayout-genericpdk  Install Klayout    │
│                             generic PDK.       │
│ layermap-to-dataclass       Converts KLayout   │
│                             LYP to a           │
│                             dataclass.         │
│ merge-gds                   Merges GDS cells   │
│                             from a directory   │
│                             into a single GDS. │
│ print-pdks                  Show installed PDK │
│                             versions.          │
│ print-plugins               Show installed     │
│                             plugin versions.   │
│ show                        Show a GDS file    │
│                             using klive.       │
│ text_from_pdf               Converts a PDF to  │
│                             text.              │
│ watch                       Filewatch a folder │
│                             for changes in     │
│                             *.py or *.pic.yml  │
│                             files.             │
│ write-cells                 Write each all     │
│                             level cells into   │
│                             separate GDS       │
│                             files.             │
╰────────────────────────────────────────────────╯
```

**To Reproduce**
Run
```
gf --help
```

**Expected behavior**
Consistent naming

tvt173, 2024-01-11:
nice. i like dashes. especially because most of the existing commands already use them

joamatab, 2024-01-11:
sounds good, lets go for dashes :-)

closed 2024-06-08 by joamatab