gdsfactory/gdsfactory #2463: Higher resolution rectangular text font

aniners, 2024-01-10:
**Is your feature request related to a problem? Please describe.**

I need to use higher resolution rectangular text fonts (https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/components/text_rectangular_font.py) or an equivalent lithography-friendly font. Currently looks like a 7x7 matrix definition of each character as opposed to 5x5 matrix should suffice. 

**Describe the solution you'd like**
Is it possible to extend/generalize text_rectangular_font.py to add higher resolution character definitions?

joamatab, 2024-01-12:
Hi Ani,

we have 3 main functions on the generic PDK for text generation

- https://gdsfactory.github.io/gdsfactory/components.html#text
- https://gdsfactory.github.io/gdsfactory/components.html#text-freetype   . this one works for any font
- https://gdsfactory.github.io/gdsfactory/components.html#text-rectangular

aniners, 2024-01-12:
I use the rectangular font defined by the matrices for each character, to be as fab-friendly as possible. Currently the matrix size for each character is 5x5, but I want to have at least 7x7 for better character polygon definition, bolding etc.

gJDelavega, 2024-01-13:
Associated PR 
Add additional & optional font to existing `text_rectangular(...)` API
#2471

joamatab, 2024-01-14:
looks great! thank you John!

closed 2024-01-14 by joamatab