gdsfactory/gdsfactory #2467: Fail to

ChineseCricket, 2024-01-12:
I tried to define a function to generate a component, and use gf.cell() function to translate this component to be a cell. To generate my component, I need to pass a class object into my function. However, I found that once I call this function, my python kernel will be shut down. Either use gf.cell(myfunction) or decorator @gf.cell to translate this function, same thing will happen. Only when I try not to pass any class object into my function it will run.

Is this a bug of gf.cell? Or it's my fault to use it wrongly? I really hope I can pass a class object into my function. I use gdsfactory in an anaconda env on Windows System with python 3.11.5. I have uploaded my code with problems in my repository mlc_dev/Cellbug.

sebastian-goeldi, 2024-01-12:
Your class needs to be json serializable as far as I can tell. 

I am getting this with your code:

```python
Traceback (most recent call last):
  File "/home/sgoeldi/repos/mlc_dev/src/First Board.py", line 88, in <module>
    LCCircuit = LCGenerator(L,C,LC,LAYER,via_pad_width)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sgoeldi/miniforge3/envs/mlc/lib/python3.12/site-packages/gdsfactory/cell.py", line 201, in wrapper
    f"{key}={clean_value_name(changed2[key])}" for key in sorted(changed.keys())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sgoeldi/miniforge3/envs/mlc/lib/python3.12/site-packages/gdsfactory/serialization.py", line 180, in clean_value_name
    return str(clean_value_json(value, fast_serialization=True))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sgoeldi/miniforge3/envs/mlc/lib/python3.12/site-packages/gdsfactory/serialization.py", line 126, in clean_value_json
    raise e
  File "/home/sgoeldi/miniforge3/envs/mlc/lib/python3.12/site-packages/gdsfactory/serialization.py", line 120, in clean_value_json
    value_json = orjson.dumps(
                 ^^^^^^^^^^^^^
TypeError: Type is not JSON serializable: CapacitorClass
```

Is there a reason to use classes? Generally it's not the best idea to have stuff live in cells, unless you really have to. As far as I can tell from your code, you are merely using the classes to store parameters?

I would suggest to either use config files or tech files which just define constants and use those instead.

If you really want to use classes, pydantic classes are json serializable afaik.

joamatab, 2024-01-12:
gf.cell is intended as a decorator of functions

See [tutorial](https://gdsfactory.github.io/gdsfactory/layout.html)

can you share the link of your repo