gdsfactory/gdsfactory #2506: component.flatten_offgrid_references()

ZhimingShi, 2024-02-04:
Hello！When I try to use 'component.flatten_offgrid_references()' command to fix the geometry which not snap to the grid, the command would change the cell's name and hierarchy, then I lost the cell's original name.  This will cause me to lose the name of the reference black box  from the fab, and thus not be able to replace the corresponding reference cell.
How can I fix the geometry which not snap to the grid and hold the original cell's name and hierarchy at the same time ?

For example, when I do not fix the geometry which not snap to the grid, there is a not complete alignment between straight and curved waveguide, and we can see  in klayout there are many independent cell.  The name called 'ligentecInvertedTaper_w3.0BB' is a black box provived by the LIGENTEC.
[image]
[image]

However，when I used 'component.flatten_offgrid_references()' command to fix the problem, the black box called ''ligentecInvertedTaper_w3.0BB'' and other cells were replaced by 'taper_new_transformed_fca64869'. This would cause the GDS failed to pass LIGENTEC's DRC check because there is no original name of the black box that they provided.
[image]

ZhimingShi, 2024-02-05:
> that's one of the limitations of black boxes, here are some options I see:
> 
> * avoid non-manhattan rotations
> * don't use black boxes
> * don't and have `flatten_offgrid_references`  and live with 1nm gaps or ask Ligentec to run `flatten_offgrid_references` on their end

Thank you for your response! If I insist on continuing to use the reference black box and not fixing this 1nm gap, will there be an effect on the waveguide etching here during the actual chip fabrication process?

tvt173, 2024-02-05:
hi @ZhimingShi , I think that question is best directed at the fab... it is likely that such a sub-nanometer gap may have no physical effect since it is so small relative to the resolution of the lithography and etching process anyways. but it depends on your process and your device, so i don't want to make assumptions

so, if it is ok with the fab, the best solution may be to turn off `flatten_offgrid_references()` and live with the slight inaccuracies and related DRC errors they might cause. otherwise, I think a true solution would require additional coordination and postprocessing on the fab's part

joamatab, 2024-02-10:
yes, as Troy said, 1nm gaps won't be able to be resolved by the litho 

however they create DRC errors that the foundry will need to avoid, which can mask other real issues if you have many of those,

Also it can appear as disconnections if you run LVS or connectivity checks