gdsfactory/gdsfactory #2513: Please update less frequently (AKA Make changes in bigger groups)

gvnwst, 2024-02-06:
I've been finding GDSFactory to be really great for my work, a substantial improvement over the more home-brewed Python layout tools I used in the past, but it feels like I've traded features for time spent debugging why my installation isn't working. 

Several times this has happened to me: 
1. Start mask layout
2. Check if there is an appropriate or nearly-appropriate factory in the generic PDK (or elsewhere)
3. Realize it's only part of a later version
4. Update GDSFactory
5. Everything is broken because it was 2 months since I updated last and there are a giant number of changes which are hard to pick out in the update manifests
6. Run down a rabbit hole of trying to fix it instead of simply slap a quick fix on the previously working version
7. Mask is delayed

Variants of this occur with "encounter a bug" and the like.

I understand that there's going to be a conflict between desire (need) to update and continuity of use. Right now I think this has swayed much too far toward frequent updates. 

I cannot admit to being a software engineer expert so this is more of a rant than a series of suggestions on what exactly to change -- though I'm happy to help develop a plan in what ways I can.

One immediate actionable way of improving this would be to lump changes in into larger groups and update minor/major versions less frequently. Right now if I want to update after 2 months, there are **TEN** releases worth of changes I need to poke through (and honestly not everything is well documented in the release readmes). It's silly to need to read through and try to understand the differences in each release and pick which one I want to update to. Packaging these changes into one or two releases would let me look at a single (larger) set of changes and decide if it's necessary.

simbilod, 2024-02-06:
This is very often brought up at the developer meetings.

One solution would be to keep the hectic update schedule on a "dev" branch, and only merge to main every so often.

Power users can work/pip install from the dev branch, and everyone else would see a slower stream of updates from the main `pip install gdsfactory` project.

gvnwst, 2024-02-06:
I'd talked with @mdecea about that, it sounded like there was some issue with dependent packages (googleX internal PDKs? Can't recall) that would be challenging to configure with a `dev` branch. 

If that sort of approach IS possible I think it would satisfy a lot of my desires.

joamatab, 2024-02-06:
Hi @gvnwst,

sorry for any inconveniences our regular updates may have caused you

why not just pin the version so you don't have to deal with this?

Do you write [tests](https://gdsfactory.github.io/gdsfactory/notebooks/08_pdk.html#testing-pdk-cells) for your code? This is the only way to make sure you can update. if you don't have tests i would strongly recommend not to update gdsfactory

We usually recommend not updating in the middle of a tapeout as we don't guarantee backwards compatibility and pin the gdsfactory in `pyproject.toml` or `requirements.txt`

i think is better to release often so people get the updates as soon as they are available, we've been releasing a couple of times a month
Most changes are documented on our [changelog](https://gdsfactory.github.io/gdsfactory/changelog.html)