gdsfactory/gdsfactory #2538: Parabolic transition

xdhuang168, 2024-02-16:
When creating parabolic transition between two cross_sections, incorrect transition is generated when the width of cross_section1 is wider than that of cross_section2. 

Workaround is possible by making sure the width in cross_section1 is narrower than that in cross_section2, and use the right port to connect to other component.

Here is the code modified from the tutorial example

```
# Create our first CrossSection
s0 = gf.Section(width=3.0, offset=0, layer=(1, 0), name="wg", port_names=("o1", "o2"))
s1 = gf.Section(width=0.5, offset=0, layer=(3, 0), name="slab")
x1 = gf.CrossSection(sections=(s0, s1))

# Create the second CrossSection that we want to transition to
s0 = gf.Section(width=0.5, offset=0, layer=(1, 0), name="wg", port_names=("o1", "o2"))
s1 = gf.Section(width=3.0, offset=0, layer=(3, 0), name="slab")
x2 = gf.CrossSection(sections=(s0, s1))

# Create the transitional CrossSection
xtrans = gf.path.transition(cross_section1=x1, cross_section2=x2, width_type="parabolic")
# Create a Path for the transitional CrossSection to follow
p3 = gf.path.straight(length=15, npoints=100)

# Use the transitional CrossSection to create a Component
straight_transition = gf.path.extrude_transition(p3, xtrans)
straight_transition.plot()
```

joamatab, 2024-10-19:
Seems to work for me

[image]

closed 2024-10-19 by joamatab

AlexDFischer, 2026-04-10:
I believe xdhuang168 was referring to the fact that parabolic tapers, if used as mode expanders or contractors for optical waveguides, should have the rapidly expanding/contracting part of the parabolic taper be the narrow part, and the slowly expanding/contracting part of the parabolic taper be the wide part. The image above does not display that behavior for both parts of the taper.

See [Optical waveguide parabolic coupling horns](https://pubs.aip.org/aip/apl/article-pdf/30/1/28/7737994/28_1_online.pdf) for the derivation of why that shape is good for optical waveguide tapers. Figure 2 is what we want.

If you're OK with this change (it may break some people's code if they were expecting the opposite behavior for some reason), I will fix it.