gdsfactory/gdsfactory #2556: clean up LayerLevel

joamatab, 2024-02-23:
from thew LayerLevel parameters  some seem useful for layout

```

        mesh_order: lower mesh order (1) will have priority over higher \
                mesh order (2) in the regions where materials overlap.
        derived_layer: Optional derived layer, used for layer_type='etch' to define the slab.
        info: simulation_info and other types of metadata.

```

others could be added into info and removed from LayerLevel

```
        width_to_z: if sidewall_angle, relative z-position (0 --> zmin, 1 --> zmin + thickness).
        z_to_bias: parametrizes shrinking/expansion of the design GDS layer \
                when extruding from zmin (0) to zmin + thickness (1).\
                Defaults no buffering [[0, 1], [0, 0]].

        layer_type: grow, etch, implant, or background.
        mode: octagon, taper, round. https://gdsfactory.github.io/klayout_pyxs/DocGrow.html
        into: etch into another layer. https://gdsfactory.github.io/klayout_pyxs/DocGrow.html
        resistivity: for metals.
        bias: in um for the etch. Can be a single number or 2 numbers (bias_x, bias_y)
        background_doping_concentration: uniform base doping level in the material (cm-3)
        background_doping_ion: uniform base doping ion in the material
        orientation: of the wafer (Miller indices of the plane)
```

simbilod, 2024-02-27:
That sounds good, although a lot of these were added because of the plugins so some refactoring will be needed

simbilod, 2024-02-27:
A lot of these were added because of the plugins. At the next developper meeting we can decide what makes sense to keep and what makes sense to only add in gplugins (probably just stuff that is useful across multiple plugins)

simbilod, 2024-02-29:
@HelgeGehring and I were discussing this yesterday. Jotting some thoughts here for general discussion

Within GDSFactory (forgetting about gplugins for now), the only use of "LayerLevels", used as dict entries in a "LayerStack", is for 3D visualization.

[Earlier versions highlighted this](
https://github.com/gdsfactory/gdsfactory/blob/78c000848474282f66421ef286b4a2e585184928/gdsfactory/technology/layer_stack.py#L8 ), with the only attributes exposed related to the geometry _associated with a single layer:_ thickness, zmin, layer, material, sidewall/z_to_bias. Everything else was in info.

Some of the things in "info" slowly moved to attributes to support recurring needs in gplugins (e.g. doping). This is easy to roll back. 

A harder thing to roll back is derived layers. The fundamental problem is that, in general, it is not true that a **single** GDS layer maps nicely to physical device levels. In general combinations of layers modify each other to generate the final "physical levels" (e.g. marker layers, etch layers, etc.).

We think we should completely get rid of the LayerLevel class in favor of a PhysicalLevel class which can take in an expression involving a combination of GDS layers (for instance (LAYER.si - LAYER.etch) + LAYER.marker -- we should overload the operators) to generate the effective layer used for the geometry. PhysicalLevel should only correspond to final geometries, and hence only support zmin, thickness, sidewall angle / z_to_bias, and mesh_order.

This is all that is needed for visualization within gdsfactory. 

Within gplugins, we should have a separate discussion about process definition (how to generate the physical LayerStack from a list of process steps associated with GDS layers, not using simulations) and modeling (physical simulation of the process, which is often overkill), which we already have some logic for.

joamatab, 2024-03-01:
Sounds good to me

We can plan this for the next major release

closed 2024-06-09 by joamatab