gdsfactory/gdsfactory #2564: CrossSection bug with auto_widen

LuYouxi, 2024-02-27:
The taper waveguide structure generated is problematic when parameters of cladding and auto_wide are present in cladding function.

import gdsfactory as gf
si_section = gf.Section(width=0.45, offset=0, layer="WG", port_names=['o1', 'o2'])
si_cld_section = gf.Section(width=10, offset=0, layer="WGCLAD")

si_wg = gf.CrossSection(
    sections=[si_section, si_cld_section], 
    radius=5, 
    bbox_layers=["WGCLAD"], 
    bbox_offsets=[0], 
    auto_widen=True, 
    auto_widen_minimum_length=50, 
    width_wide= 1,
)

c = gf.Component("sample_connect")
mmi1 = c << gf.components.mmi1x2()
mmi2 = c << gf.components.mmi1x2()
mmi2.move((100, 50))
route = gf.routing.get_route(mmi1.ports["o2"], mmi2.ports["o1"], cross_section=si_wg)
c.add(route.references)
c.show()

[image]

In past versions, the output of this script looked like this：
[image]

closed 2024-02-27 by joamatab

joamatab, 2024-02-27:
Thank you for reporting this,

can you try `pip install gdsfactory --upgrade`?

[image]

this should be fixed in the latest gdsfactory release 7.14.0

LuYouxi, 2024-02-28:
> Thank you for reporting this,
>
> can you try `pip install gdsfactory --upgrade`? [...]

It has been repaired, thank you!

PR #2565 (2024-02-27): fix taper extrude
fixes #2564 

[image]