gdsfactory/gdsfactory #2569: .to_yaml() misses some layer properties defined in Klayout

smgitrep, 2024-02-28:
I created a lyp file in Klayout and converted it to a yaml file using .to_yaml(), but the resulting yaml file does not include all layer properties.  For example, it leaves out the "stiple" pattern and line "style" defined in Klayout.

joamatab, 2024-03-05:
yes, the LYP misses some layer properties

PRs are welcome

@thomasdorch

thomasdorch, 2024-03-05:
Originally I wrote the whole LayerView processing stuff to avoid using klayout (it was an optional dependency at the time) but now that kfactory/klayout are part of the requirements this should be refactored to use the kfactory lyp import/export.

smgitrep, 2024-03-05:
Thank you for the comments.  I decided to setup layerviews using a yaml file, and then, convert it to lyp for use in Klayout.  This seems easier.  I found hatch (Klayout "stiple") and line styles (Klayout "style") in one of the python scripts included with gdsfactory.  See attached txt file.  I also found colors can be set using color names (strings) in the yaml file and to_lyp converts them to hex values for use in Klayout.

[klayout hatch and line styles.txt](https://github.com/gdsfactory/gdsfactory/files/14501607/klayout.hatch.and.line.styles.txt)

joamatab, 2024-04-11:
@skychil  also was working on a CSV version of this