gdsfactory/gdsfactory #2600: routing bug:  start_straight_length does not create a start straight section

smgitrep, 2024-03-07:
Possible bug:  gf.route.get_route() does not create a "start straight length"

Expected behavior:  I expected a straight waveguide at the output of mmi1 since I set start_straight_length = 10.

Code:

new_strip = partial(
    gf.cross_section.cross_section,
    width=0.5,
)

c = gf.Component("sample_connect")
mmi1 = c << gf.components.mmi1x2()
mmi2 = c << gf.components.mmi1x2()
mmi2.move((500, 100))

bend = partial(gf.components.bend_euler, cross_section=new_strip)
straight = partial(gf.components.straight, cross_section=new_strip)

route = gf.routing.get_route(
    mmi1.ports["o3"],
    mmi2.ports["o1"],
    bend=bend,
    straight=straight,
    auto_widen=True,
    start_straight_length=10,
    end_straight_length=10,
    width_wide=2,
    auto_widen_minimum_length=50,
    taper_length = 10,
    radius=30,
    cross_section=new_strip,
)
c.add(route.references)

c.show()

joamatab, 2024-03-07:
Hi Sagi,

the start straight is a minimum length when starting the route but is not guaranteed

smgitrep, 2024-03-07:
Ok. Maybe that should be mentioned in the tutorial?