gdsfactory/gdsfactory #2609: no more info and post_process

joamatab, 2024-03-08:
How about using UnPack instead of partial for customizing PDKs?

https://typing.readthedocs.io/en/latest/spec/callables.html#unpack-kwargs

[image]

```python
import gdsfactory as gf
from typing import Unpack

@gf.cell(post_process=[gf.add_padding, gf.add_pins.add_pins])
def straight_sc(**kwargs: Unpack[gf.c.Straight]):
    return gf.components.straight(**kwargs)

if __name__ == "__main__":
    c = straight_sc()
    c.show()
```

@nikosavola 
@sebastian-goeldi 
@tvt173 
@flaport

nikosavola, 2024-03-09:
Seems like a nice way to type components based on other components.

I wonder would it be easy to generate the TypedDict from a component function automatically, just by looking at which keyword arguments have defaults available and applying [`NotRequired`](https://docs.python.org/3/library/typing.html#typing.NotRequired) to those.

sebastian-goeldi, 2024-03-09:
One caveat through: Unpack seems to only be available in python 3.12+ for kwargs [PEP692](https://peps.python.org/pep-0692/]

joamatab, 2024-03-09:
It worked for me on 3.11

joamatab, 2024-03-11:
I think this is a better solution

https://github.com/gdsfactory/gdsfactory/pull/2618