gdsfactory/gdsfactory #2627: Issue on page /api_routing.html all_angle_ routing

jsolerOnet, 2024-03-16:
In the code attached I try to route from a imported gds BB (an AMF PBRS) to a silicon edge coupler that is tilted so I am trying to use:
routes = gf.routing.get_bundle_all_angle([Out_ARR[lev].ports["o1"]], [PBRS_ARR[63-15].ports["o1"]],cross_section=Si_crossection,
                                                      bend=gf.components.bend_euler_s,end_cross_section=Si_crossection) 

Initially I was getting an error with the layers not matching (layer (10,0) for the rib in AMF and layer (1,0) for the waveguide layer in the standard PDK. However I changed the layer map to do this, assigning layer (10,0) to WG. Once I did this the previous issue went away but now I am getting the following error:

2024-03-16 02:50:22.654 | WARNING  | gdsfactory.routing.all_angle:_make_error_trace:204 - RouteWarning: Not enough room to route between ports: {'name': 'o2', 'width': 0.5, 'center': [6887.781103, 11462.115667], 'orientation': 162.48, 'layer': [10, 0], 'port_type': 'optical'} and {'name': 'o2', 'width': 0.5, 'center': [5560.742591, 7077.156797], 'orientation': 0.0, 'layer': [10, 0], 'port_type': 'optical'}
C:\Users\JordiSolerPenades\anaconda3\Lib\site-packages\gdsfactory\routing\all_angle.py:204: RouteWarning: Not enough room to route between ports: {'name': 'o2', 'width': 0.5, 'center': [6887.781103, 11462.115667], 'orientation': 162.48, 'layer': [10, 0], 'port_type': 'optical'} and {'name': 'o2', 'width': 0.5, 'center': [5560.742591, 7077.156797], 'orientation': 0.0, 'layer': [10, 0], 'port_type': 'optical'}
  warnings.warn(message, RouteWarning)
Traceback (most recent call last):
  File "C:\Users\JordiSolerPenades\anaconda3\Lib\site-packages\gdsfactory\component.py", line 1334, in add
    self.add(subelement)
  File "C:\Users\JordiSolerPenades\anaconda3\Lib\site-packages\gdsfactory\component.py", line 1334, in add
    self.add(subelement)
  File "C:\Users\JordiSolerPenades\anaconda3\Lib\site-packages\gdsfactory\component.py", line 1334, in add
    self.add(subelement)
  [Previous line repeated 995 more times]
  File "C:\Users\JordiSolerPenades\anaconda3\Lib\site-packages\gdsfactory\component.py", line 1332, in add
    elif isinstance(element, Iterable):
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen abc>", line 119, in __instancecheck__
RecursionError: maximum recursion depth exceeded in comparison

This still happens if I separate the blocks more since from the traceback it seemed there might be an issue with the blocks being to close but there seems to be another issue that I don't know how to fix.

jsolerOnet, 2024-03-18:
Just to clarify for everyone we managed to sort this out:
The main issue was from get_bundle_all_angle, since it is a bundle routing function it was expecting lists so I just added square brackets around the ports definition and that made them 1 element lists but what I was not expecting was that the function was also returning a 1 element list so when I added the reference I was adding a list instead of a route and resulted into the very long loop that was giving the error.
To solve this issue the solution was to just reference the first and only element of the list, that simple.

tvt173, 2024-03-19:
great, sorry for the confusion. glad you were able to sort this out

closed 2024-03-19 by tvt173