gdsfactory/gdsfactory #2673: how to create two independent cells in gdsfactory?

wuzhonghan, 2024-04-15:
Hi,

I have a question：How to create two independent cells in gdsfactory?

in GDSTK, I can create two cells like below:

```
import gdstk

lib = gdstk.Library()

cell1 = lib.new_cell("FIRST")
cell2 = lib.new_cell("SECOND")

rect = gdstk.rectangle((0, 0), (2, 1))
cell1.add(rect)
cell2.add(rect)
```
[image]

How can I create two cells or more like this using gdsfactory?

joamatab, 2024-04-15:
very similar in gdsfactory,

the main difference is that we always create a top cell

```python

import gdsfactory as gf

c = gf.Component('top')
cell1 = gf.Component("FIRST")
cell2 = gf.Component("SECOND")

rect = gf.c.rectangle((0, 0), (2, 1))
cell1.add_ref(rect)
cell2.add_ref(rect)
c.show()
````

wuzhonghan, 2024-04-16:
> very similar in gdsfactory,
> 
> the main difference is that we always create a top cell
> 
> ```python
> import gdsfactory as gf
> 
> c = gf.Component('top')
> cell1 = gf.Component("FIRST")
> cell2 = gf.Component("SECOND")
> 
> rect = gf.components.rectangle(size=(2, 1))
> cell1.add_ref(rect)
> cell2.add_ref(rect)
> c.show()
> ```

Thanks, do you mean that I can only create cells like this?
[image]

joamatab, 2024-04-16:
yes

closed 2024-04-17 by wuzhonghan