gdsfactory/gdsfactory #2684: gdsfactory.fill_rectangle has some bugs

wuzhonghan, 2024-04-17:
**The bug**
Dummy fill in gdsfactory is not as expect. 
Recently, I want to use the dummy fill function in gdsfactory such as **gdsfactory.fill_rectangle**, but I find that the avoided layers do not work as expect. Could any one fixed this problem?

**To Reproduce**
```
import gdsfactory as gf

coupler_lengths = [10, 20, 30, 40, 50, 60, 70, 80]
coupler_gaps = [0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8]
delta_lengths = [10, 100, 200, 300, 400, 500, 500]

mzi = gf.components.mzi_lattice(
    coupler_lengths=coupler_lengths,
    coupler_gaps=coupler_gaps,
    delta_lengths=delta_lengths,
)

# Add fill
c = gf.Component("component_with_fill")
layers = [(1, 0)]
fill_size = [4, 4]

c << gf.fill_rectangle(
    mzi,
    fill_size=fill_size,
    fill_layers=[(2, 0)],
    margin=0,
    fill_densities=[0.8] * len(layers),
    avoid_layers=layers,
)

c << mzi
c.show()

```
[image]
Look at the rightest MZI, you will find that the rectangle filling pattern intersect with the waveguide, which is not I expect.

How could we fix the bug?

joamatab, 2024-04-17:
fill_rectangle is deprecated and will be removed soon, use gf.geometry.fill_klayout instead

wuzhonghan, 2024-04-18:
thank you guys!

wuzhonghan, 2024-04-18:
@joamatab  @amccaugh  Hi guys! May I ask two more questions?
1. As for gf.geometry.fill_klayout, can I use custom filling pattern, such as a ring shape.
2. As for gf.geometry.fill_klayout, does it support enhanced fill feature like in klayout software?
[image]

joamatab, 2024-04-18:
Hi,

1. yes you can take a look at the [docs](https://gdsfactory.github.io/gdsfactory/notebooks/04_components_geometry.html#dummy-fill-tiling)
2. i don't know, we are using kfactory fill tool, this is more of a kfactory or klayout question

wuzhonghan, 2024-04-19:
Thanks a lot!

closed 2024-04-19 by wuzhonghan