gdsfactory/gdsfactory #2755: PyCon 2024 sprint ideas

joamatab, 2024-05-19:
PyCon sprints for gdsfactory will be from Monday 19 to Thursday May 23rd at the Pittsburgh Convention Center

## Getting started

- [ ] improve on boarding experience to the library (docs, installation ...)
- [ ] Increase test coverage
- [ ] Make mypy pass
- [ ] Review docstrints
- [ ] Try to do something with the library (chip design, 3D printing, PCB layout ...)

## More advanced topics

- [ ] improve [extensions/plugins](https://github.com/gdsfactory/gplugins)
- [ ] integration with SPICE and Open source chip libraries ([skywater130](https://github.com/dan-fritchman/Hdl21/tree/main/pdks/Sky130/sky130_hdl21))

@sebastian-goeldi 
@dan-fritchman
@tvt173 
@flaport 
@Vivswan

dan-fritchman, 2024-05-20:
@joamatab, do you have any thoughts on what you'd want to achieve in that "advanced" section? 

I won't be at the conference, but if you've got thoughts on *what* you want, I can probably write up some guidance on *how*.

joamatab, 2024-05-20:
How about linking the [skywater](https://github.com/gdsfactory/skywater130) layout generators to the SPICE models from hdl21?

@proppy

dan-fritchman, 2024-05-24:
Sounds great.  
https://github.com/dan-fritchman/Hdl21/issues/220 should be a helpful enabler for ya there!

daquintero, 2024-05-27:
Thanks a lot Dan this is really helpful!