gdsfactory/gdsfactory #2775: fix kfactory dependency version

tvt173, 2024-05-22:
**Describe the bug**
The version requirements for kfactory are too loose. When installing gdsfactory with an old version of kfactory already installed, gdsfactory will fail on load.

**To Reproduce**
Try installing gdsfactory after installing kfactory 0.12 or lower.

**Expected behavior**
gdsfactory runs without issue

**Suggested fix**
Update the requirements... maybe `"kfactory[git,ipy]>=0.13.0,<0.14"` ?

@sebastian-goeldi
@joamatab

joamatab, 2024-05-22:
Yes, we need a new kfactory release

sebastian-goeldi, 2024-05-22:
I think I suggested @joamatab to use the following until we are close to release
`kfactory @ git+https://github.com/gdsfactory/kfactory.git#main`

joamatab, 2024-05-31:
yes, we'll need to release a new kfactory before this gdsfactory release

joamatab, 2024-06-02:
this should be fixed

closed 2024-06-02 by joamatab