gdsfactory/gdsfactory #2787: fix routing

joamatab, 2024-05-26:
some routing cases are still broken

[image]

@sebastian-goeldi 
@tvt173

sebastian-goeldi, 2024-05-26:
Is the right one 4 (or at least 2) separte calls to `route_bundle` or a single one?

If it's a single one, it will not work as the end ports must be a single bank of ports and not overlap with the starting bounding box.

That would explain why it doesn't work.

I also noticed that routing fails if the routes are exactly lining up perfectly behind each other. E.g. both end ports are angle=0 and end port has same `y` as start port. I suspect the router fails to pick a side and doesn't do anything, which causes it to then fail horribly when it picks the next step, because the previous routes aren't where they are supposed to be.

sebastian-goeldi, 2024-05-26:
Fixed by https://github.com/gdsfactory/kfactory/pull/337

closed 2024-05-31 by joamatab