gdsfactory/gdsfactory #2797: how to add virtual cells

joamatab, 2024-05-29:
```python

"""Snap bends together."""

from __future__ import annotations

import gdsfactory as gf

@gf.cell(check_instances="vinstances")
def snap_bends() -> gf.Component:
    c = gf.Component("snap_bends")
    b1 = c << gf.c.bend_euler()
    b1.rotate(37)
    b2 = c << gf.c.bend_euler()
    b2.connect("o1", b1.ports["o2"])
    return c

if __name__ == "__main__":
    c = snap_bends()
    c.show()

```

@sebastian-goeldi 
@tvt173

tvt173, 2024-05-29:
for clarity, this runs, but there is a gap at the connection, so it does not appear to be behaving correctly

sebastian-goeldi, 2024-05-29:
Have a look at `kfactory.cells.virtual` and pick one of them or all`

sebastian-goeldi, 2024-05-29:
I woupd suggest to have them all in gdsfactory as well. You need to add virtual cells for anything that's non-manhattan, so, straights and non-90° bends

tvt173, 2024-05-29:
so is the expectation that a regular cell placed as a virtual instance will still have snapping errors like this?

sebastian-goeldi, 2024-05-29:
That's not an expectation, that will always be the case. Problem is a normal cell will always have angle errors on non-90° ports (realistically).

As soon as you insert shapes (such as polygons) they get snapped. You can add or retrieve them as um based instances such ass `shape.dpolygon`, nevertheless, they are snapped.

Therefore, you then are left with two terrible choices. 1) take the gap hit (angle error between port angle and physical shape) or 2) take an angle hit (i.e. correct for the snapping error in the angle)

Both aren't really acceptable. Thus the concept of VKCells or VInstance which defer the snapping until inserted into a KCell.

sebastian-goeldi, 2024-05-29:
The reason for straights being needed as a virtual cell is much less that you can't use it, but rather that if you use the all_angle router, a standard straight can still cause problems because you are snapping its shapes twice. Might be ok in a lot of cases, but less so in the routing case.

tvt173, 2024-05-29:
actually, this works

```python
"""Snap bends together."""

from __future__ import annotations

import gdsfactory as gf

@gf.cell(check_instances="vinstances")
def snap_bends() -> gf.Component:
    c = gf.Component("snap_bends")
    b1 = c.create_vinst(gf.c.bend_euler())
    b2 = c.create_vinst(gf.c.bend_euler())
    b1.rotate(37)
    # b1 = c << gf.c.bend_euler()
    # b1.rotate(37)
    # b2 = c << gf.c.bend_euler()
    b2.connect("o1", b1.ports["o2"])
    return c

if __name__ == "__main__":
    c = snap_bends()
    c.show()
```

so i think it's more a matter that you are calling the `rotate()` method on the normal instance before it gets converted to a regular one. when creating as a virtual instance from the start, it works

tvt173, 2024-05-29:
when the underlying cells already have snapping errors introduced, i.e. by creating an euler bend with non-90 degree angle, as shown below, then you get issues

```python
"""Snap bends together."""

from __future__ import annotations

import gdsfactory as gf
from kfactory.cells.virtual.euler import virtual_bend_euler

@gf.cell(check_instances="vinstances")
def snap_bends() -> gf.Component:
    c = gf.Component()
    b1 = c.create_vinst(gf.c.bend_euler(angle=25))
    b2 = c.create_vinst(gf.c.bend_euler(angle=39))
    b1.rotate(37)
    b2.connect("o1", b1.ports["o2"])
    return c

if __name__ == "__main__":
    c = snap_bends()
    c.show()
```

but there does not seem to be an issue when placing normal cells without snapping issues as virtual instances with off-angle transformations

tvt173, 2024-05-29:
also this code gives an error

```python
"""Snap bends together."""

from __future__ import annotations

import gdsfactory as gf
from kfactory.cells.virtual.euler import virtual_bend_euler
from kfactory.kcell import vcell

@vcell()
def snap_bends() -> gf.Component:
    c = gf.Component()
    b1 = c.create_vinst(gf.c.bend_euler(angle=29))
    b2 = c.create_vinst(gf.c.bend_euler(angle=39))
    b1.rotate(37)
    b2.connect("o1", b1.ports["o2"])
    return c

if __name__ == "__main__":
    c = snap_bends()
    c.show()
```

[image]

joamatab, 2024-05-29:
how about having a `gf.ComponentOffgrid`?

sebastian-goeldi, 2024-05-29:
> how about having a `gf.ComponentOffgrid`?

I would make intent or purpose a bit clearer. Maybe `ComponentAllAngle`? Or shorten it

joamatab, 2024-05-29:
yes, ComponentAllAngle sounds good to me

closed 2024-05-31 by joamatab