gdsfactory/gdsfactory #2821: fix plot_netlist(recursive=True)

joamatab, 2024-06-04:
**Describe the bug**
plot_netlist does not work well when recursive=True

**To Reproduce**
```
    import matplotlib.pyplot as plt

    import gdsfactory as gf

    cpl = (10, 20, 30, 40)
    cpg = (0.2, 0.3, 0.5, 0.5)
    dl0 = (0, 50, 100)

    c = gf.c.mzi_lattice(
        coupler_lengths=cpl, coupler_gaps=cpg, delta_lengths=dl0, length_x=1
    )
    n = c.get_netlist(recursive=True)
    c.plot_netlist(recursive=True)
    plt.show()
```

[image]

joamatab, 2024-06-04:
this is how it should look like

[image]

@tvt173 
@nikosavola

simbilod, 2024-06-04:
> this is how it should look like
> 
> ![Image](https://private-user-images.githubusercontent.com/4514346/336618148-85530e23-f353-4e22-a5d9-ea718674d837.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1Mzk1NjIsIm5iZiI6MTcxNzUzOTI2MiwicGF0aCI6Ii80NTE0MzQ2LzMzNjYxODE0OC04NTUzMGUyMy1mMzUzLTRlMjItYTVkOS1lYTcxODY3NGQ4MzcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDRUMjIxNDIyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2UzMDAxMDhkZDcyMWZiMDEyMzhjOGIyYTRhMmRiNzU5ZmIxMGQ1YzU5ZmRmNDE1NmNlZjJlYTBjNzdlNzk5MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Wx_UzJmQ5Ti6iTpMXGh16e6Ew29chODVLvBzcCS5vG8)
> 
> https://gdsfactory.github.io/gplugins/notebooks/11_get_netlist.html
> 
> @tvt173 @nikosavola @simbilod

that's get_netlist_flat which has been removed

simbilod, 2024-06-04:
I'll help you migrate `get_netlist_flat` if you help me migrate `klayout_3D_script` with the new LayerStack over at #2805  ;)