gdsfactory/gdsfactory #2832: Refactor core functions from `@cell`-decorated containers

tvt173, 2024-06-07:
**Is your feature request related to a problem? Please describe.**
Functions which take a cell, somehow decorate them (by adding grating couplers, straights, pins, etc.), and return a new, decorated cell we often refer to as "containers". These functions, in principle, should be generic, but by registering them with `@cell` we introduce some problems:
- they must be registered in downstream PDKs too for proper functionality
- they introduce an extra layer of hierarchy which is not always desired

It would be nice to refactor out the core functions so that they could be used more flexibly.

For example 

```python
@cell
def add_grating_couplers(component, ...):
    c = gf.Component()
    # add component and other stuff to c
    return c
```

Would become
```python
def add_grating_couplers(top_component, inner_component, ...):
   # add inner_component and other stuff to top_component
   return top_component
```

Note that the current `@cell` containers could still exist and simply be refactored to use the new, generic functions.

@flaport 
@sebastian-goeldi 
@joamatab

joamatab, 2024-06-08:
How can we do this and don't create duplicated cell names, when, for example, using a file watcher?

sebastian-goeldi, 2024-06-09:
Why would that matter? This would only impact how the general workflow is. If your cell wants pure instances, this would just recreate that cell. This change would allow you to do the following (the add_gc might not be the best example, but e.g. add_extents is a good example)
```python
@cell
def my_coupled_component() -> gf.Component:
    c = gf.Component()
    my_comp_ref1 = c << my_comp
    add_grating_coupler(my_comp_ref1, gc_array)
    my_comp_ref2 = c << my_comp
    add_grating_coupler(my_comp_ref2, gc_array)
    my_comp_ref3 = c << my_comp
    add_grating_coupler(my_comp_re3, gc_array)
    return c
```

Without unecessarily creating a hierarchy level for each ref or having to have an intermediate level where it might not be desired

tvt173, 2024-06-10:
yes exactly, it doesn't matter. these are functions that should be used _within_ other cell functions

I think you mean for your second argument to be `c`, not `gc_array`, right @sebastian-goeldi ? or I would suggest as the first argument like this

```python
@cell
def my_coupled_component() -> gf.Component:
    c = gf.Component()
    my_comp_ref1 = c << my_comp
    add_grating_coupler(c, my_comp_ref1)
    my_comp_ref2 = c << my_comp
    add_grating_coupler(c, my_comp_ref2)
    my_comp_ref3 = c << my_comp
    add_grating_coupler(c, my_comp_ref3)
    return c
```

sebastian-goeldi, 2024-06-10:
Ah, yes of course. I mixed my mind a bit up when using the function as I barely use wrapper functions ^^