gdsfactory/gdsfactory #2841: Upgraded layerstack further improvements

simbilod, 2024-06-08:
Following up on #2805 

* Fix 3D KLayout script. We can convert the booleans of the derived layers to KLayout booleans directly.
* Buffering of layers as operations, for instance 
  * layer1 + 2 can buffer the polygons of layer1 by 2um
  * layer2 * 2 can buffer the polygons of layer2 by a factor of 2
  * layer1 - 2 to shrink would need to separate out the "-" used to NOT another layer from integer

Collect other ideas in this issue!

sebastian-goeldi, 2024-06-09:
Could you elaborate on this:

> layer1 - 2 to shrink would need to separate out the "-" used to NOT another layer from float for shrinking

why is this different than `+`?

simbilod, 2024-06-10:
> Could you elaborate on this:
> 
> > layer1 - 2 to shrink would need to separate out the "-" used to NOT another layer from float for shrinking
> 
> why is this different than `+`?

The `__sub__ `operator is currently overloaded to accept another layer to perform boolean difference. 

I think we will have to do something of this sort:

```
def __sub__(self, other):
    if isinstance(other, AbstractLayer):
        <return DerivedLayer with the NOT operation applied>
    elif isinstance(other, int) or isinstance(other, float):
        <return a new AbstractLayer with the buffer attribute set>
```

It's possible that `layer + (-number)` would work the way you describe, however.

sebastian-goeldi, 2024-06-10:
I see. Any reason to have `+` not be OR then?

I would suggest with sizing use a function then, something akin to klayout e.g. `layer1.sized(2)`

simbilod, 2024-06-11:
Yes currently either `|` or `+` can be used for OR.

You are right that only having methods for sizing would make this less confusing