gdsfactory/gdsfactory #2843: When I output the script using gdsfactory.generic_tech.get_klayout_pyxs, I find that it does not work properly in klayout. The high probability of initial debug discovery is related to the deposit() function. The error message for klayout is as follows:

joamatab, 2024-06-09:
When I output the script using gdsfactory.generic_tech.get_klayout_pyxs, I find that it does not work properly in klayout. The high probability of initial debug discovery is related to the deposit() function. The error message for klayout is as follows:
[image]
How to solve this problem?

_Originally posted by @LuYouxi in https://github.com/gdsfactory/gdsfactory/discussions/2842_

sebastian-goeldi, 2024-06-09:
This is most likely an issue of a polygon constructor not used correctly. I would suggest to either use one with named arguments (still not 100% guaranteed but better success rate) or check the error in the macros editor

LuYouxi, 2024-06-09:
> This is most likely an issue of a polygon constructor not used correctly. I would suggest to either use one with named arguments (still not 100% guaranteed but better success rate) or check the error in the macros editor

I checked the following error message in the macros editor:
[image]

joamatab, 2024-06-12:
Hi LuYouxi,

i found a similar issue when adding a Polygon with 4 points in klayout, that was casted to a Box. See [issue](https://github.com/KLayout/klayout/issues/1651)

@klayoutmatthias

joamatab, 2024-06-12:
it seems like that error was fixed, 

which version of klayout are you using?

LuYouxi, 2024-06-13:
> it seems like that error was fixed,
> 
> which version of klayout are you using?

I used klayout version 0.29.1.

LuYouxi, 2024-06-15:
> > it seems like that error was fixed,
> > which version of klayout are you using?
> 
> I used klayout version 0.29.1.

sorry, The version information is wrong. It should be 0.28.14.

sebastian-goeldi, 2024-06-15:
Please update klayout to 0.29+ then. I believe there were some API changes which may made the plugin incompatible. And also make sure your plugin is on the newest version (open the package manager and see the update tab)

klayoutmatthias, 2024-06-15:
Yes, the error message is related to https://github.com/KLayout/klayout/issues/1651. Updating to 0.29.x should fix the problem. There was an API change, but it was not intended. It is a side effect of https://github.com/KLayout/klayout/issues/1530, released with at 0.28.14.

Matthias

joamatab, 2024-10-19:
Im moving this issue to the related repo

https://github.com/gdsfactory/klayout_pyxs/issues/11

closed 2024-10-19 by joamatab