gdsfactory/gdsfactory #2844: Corner Rounding, Offset and Outline Operations

Alessio-Zicoschi, 2024-06-09:
Hello, it would be nice to have the operations of filleting, offset and boolean implemented on gdsfactory components. These were present in previous versions of gdsfactory. I think they have been removed in v8. 
- Filleting was implemented through gdstk polygons, but in klayout there should be an analogous function. I could not find anything in v8.
- The offset operation is mentioned in the component tutorial through the p1.sized() function, but the implementation is pretty cumbersome.
- The outline operation was present in previous versions and is just the combination of offset and not operations. It is not crucial but I found it neat to use.
It would be nice to have these operations directly acting on gdsfactory components. Even if I have to deal with simple polygons, I find it much better to use gdsfactory components rather than moving to gf.kdb.DPolygon. Thanks in advance!

sebastian-goeldi, 2024-06-09:
Fillet is implemented with `(D)Polygon.round_corners` https://klayout.de/doc/code/class_DPolygon.html#method57

The reason the offset is not as easy is because imo gdstk uses rather restrictive assumptions for getting the outline.
How do you size? How do you handle sharp corners? How do you handle self-intersecting polygons? And so on.

I think this would be quite hard to define a sane default behavior and generally requires individual adjustments per component.

but generally an outline (of a layer) can happen two ways:

```python

r = gf.kdb.Region(c.shapes(layer_index)))
c.shapes(layer_index).insert(r.sized(outline_in_dbu)-r) # not r.sized vs r.size; the latter will change in place and therefore not work
```
if it should be non-recursive

if recursive
```python

r = gf.kdb.Region(c.begin_shapes_rec(layer_index)))
c.shapes(layer_index).insert(r.sized(outline_in_dbu)-r)
```

This is the standard isotropic sizing case. Note this can cause problems with self-intersecting polygons. For safe usage a minkowski sum is most likely necessary.

tvt173, 2024-06-10:
maybe it would be useful to add functions that are compatible with the old behavior though? and we can add options to customize as necessary?

joamatab, 2024-06-10:
Yes, I apologize for the documentation not being yet as extensive as it was in version 7.

However, these operations are now much easier due to the enhanced functionality and usability of the Klayout classes.

I can some examples on how to do this

joamatab, 2024-06-10:
We added some examples in https://github.com/gdsfactory/gdsfactory/pull/2845

let us know if you find further improvements

closed 2024-06-10 by joamatab

Alessio-Zicoschi, 2024-06-11:
Hello, thanks for the explanations and for adding the examples to the tutorials, I found them pretty useful and they allowed me to do what I wanted. I also understand the concerns about offsetting, but it is very useful to have a simple way to handle non-problematic cases, as shown in the tutorial. Just a couple more questions:
1. Is there a possibility to round different corners of a polygon with different radii? I could not find this functionality in `(D)Polygon.round_corners`, I was wondering if there is any possibility to implement it.
2. It seems to me that booleans and `grid` function do not have the cell decorator anymore, thus they produce unnamed components. Am I missing something? Is this made on purpose?
Thanks a lot in advance!

sebastian-goeldi, 2024-06-12:
Hello @Alessio-Zicoschi,

Klayout does support that natively, no. I onlu supports rounding for inner vs outer radii (I think defined as `<180°` opening vs bigger).

If you want more rounding options, I am afraid you'll either have to cut out the pieces you want manually or export and afterwards import to something like shapely.

If you absolutely want to stay in KLayout, I guess you could try to create partial paths and then round them and then stitch them back together. But that seems like a lot of work for what other packages can do better most likely