gdsfactory/gdsfactory #2847: gdsfactory8 issues

lokgar, 2024-06-10:
**Issues Reproducing Routing Examples for Fiber Arrays/Pads**
[image]

Recent routing changes have caused code failures when reproducing examples for fiber arrays and pads. The solution is not immediately clear. Is the documentation not yet updated or there are issues with the release?

**KLive Behavior Changes on Windows**
KLive now saves temporary GDS files for KLayout three levels up from the directory where the script or notebook is run. This is different from the previous behavior of saving them to the temporary folder in the home directory (e.g., AppData/Local/Tmp). The new behavior is inconvenient and confusing.

sebastian-goeldi, 2024-06-10:
For the saving pattern, it creates a build folder in your git repository if there is one (and doesn't clean up).
If it's not a git repository, it will create a temporary file and clean it up. What different behavior would you expect for the save?

Having a build folder is kind of standard in a lot of programming languages.

@tvt173 maybe we shouldn't have added that check :smile:

lokgar, 2024-06-10:
Hi @sebastian-goeldi,

I'm referring to the show() function within KLive, not saving files directly. When I experiment with components in notebooks, my build folder gets cluttered with temporary GDS files. This doesn't happen in gdsfactory7, which uses the PC's temporary directory instead. It's not a crucial problem for sure, I've just noticed the difference (which is a bit annoying) between gdsfactory7 and gdsfactory8 release.

sebastian-goeldi, 2024-06-10:
I get that. It's a problem with ipython (or rather I didn't expect this when I wrote the logic for show). Just to clarify this doesn't have anything to do with klive. klive only loads whatever it gets instructed to load with a simple HTTP request containing a JSON.

The show logic changed because gdsfactory is now using kfactory as a backend. And in the end it uses `stem("file.py").oas` as the name, but somehow in ipython that resolves to a number which I suspect changes with every change in the notebook.

I'll look into it in kfactory. In the meantime the easy solution is (please make absolutely sure all files you want to have tracked are in `git status` is to run `git clean -xdf` in the root folder; this will recursively delete all unknown (untracked and ignored & unknown) files for git.

Alternatively just delete the build folder some what regularly.

I will try to fix this in the show behavior of kfactory and try to get a consistent name for the oas/gds.

sebastian-goeldi, 2024-06-10:
I can't reproduce this with a standard cpython kernel though. I can in an ipython kernel, but not the standard python.

sebastian-goeldi, 2024-06-10:
Ah ok, but then it is working as intended. It creates one file per `*.py` file that is executed.

With jupyter the problem is it will create `13527282.oas` and the number keeps changing

treep2, 2024-06-13:
What happened to the ports?

Just upgraded from an older version (5 haha, had a lot of custom stuff), suddenly I am not able to use any tapers anymore. I keep getting stuck with:
"dwidth needs to be even to snap to grid. Got 0.155"

I've been digging through the code and it happens when the taper tries to generate its ports. Why is this function forcing me to change my taper width to an even number?

sebastian-goeldi, 2024-06-13:
The problem is there is no good solution to this.

Assuming you are on a 1nm dbu proccess, a taper with 155nm width would have the port center at 77.5nm, therefore off-grid. Similar a wavguide with 155nm width cannot have an on-grid center, or if it is, the wavegide sides are off-grid, which is even worse.

This can cause 1nm shifts because depending how you are rounding and positioning, either you port would be assymetric wrt the center or your center is not on grid.

Sure there would be ways to calculate ports which are off-grid, but the design of it to safely avoid snapping errors require a lot of overhead which is just not worth the headache.

I would suggest to move to either 154 or 156nm width.

If you really are in a regime where it would matter foe the accuracy you probably should request your foundry to go below 1nm dbu

tvt173, 2024-06-13:
@sebastian-goeldi , i agree that probably the best solution in this case would be to avoid off-grid features. but if we just assume for a moment that @treep2 has a justifiable need for this, shouldn't he be able to do it with virtual cells and instances?

sebastian-goeldi, 2024-06-13:
> @sebastian-goeldi , i agree that probably the best solution in this case would be to avoid off-grid features. but if we just assume for a moment that @treep2 has a justifiable need for this, shouldn't he be able to do it with virtual cells and instances?

Correct. With virtual cells and instances it is possible to have this. I actually meant to add this in my comment but forgot ;). However, I believe gdsfactory at them moment doesn't have the virtual taper function, yet.

joamatab, 2025-03-05:
Closing this for now Denis,

Feel free to reopen

closed 2025-03-05 by joamatab