gdsfactory/gdsfactory #2850: make gdsfactory klayout package to depend on klive and meta_info_ports

joamatab, 2024-06-10:


sebastian-goeldi, 2024-06-10:
does this even still work? what is the gdsfactory klayout package doing?

tvt173, 2024-06-10:
the point of this is as we discussed earlier: so that users can get all the gdsfactory-related tools with one package install in klayout

i don't particularly care how it's done, but if there's already a "gdsfactory" package, i think it would make sense that it is the go-to one that has everything. it will be the easiest to find.

i think Joaquin said that it currently installs just the generic tech and maybe a few other things

joamatab, 2024-06-11:
yes, at the moment the gdsfactory package only installs the layermap

i can take care of this issue

sebastian-goeldi, 2024-06-11:
Oh. If it's just that and not the full bundle, just add `klive` and `metainfo_ports` as dependencies to the package

closed 2024-06-11 by joamatab