gdsfactory/gdsfactory #2855: Reintegrate  c.plot_matplotlib()

ZhimingShi, 2024-06-11:
Hello! After updating gdsfactory version 8.0, I found that the c.plot_matplotlib() command has been removed as unavailable. It is highly recommended to re-add this function, as it is particularly handy to use it to see the dimensions of the plotted figure and the coordinates where it is located. What you see is what you get, the other c.plot() and c.show() commands do not allow you to see the exact dimensions and centre coordinates of the drawing!

joamatab, 2024-06-11:
why not use klayout?

```
c = gf.components.straight(0
c.show()
```

the current plot has a scale bar
[image]

my main issue with c.plot_matplotlib() was that it was always too slow for complex layouts

johnyb115, 2024-06-12:
To hijack this, how do you show ports in Klayout now? the c.show() does not accept the show_ports=True anymore.

sebastian-goeldi, 2024-06-12:
Built into klayout now since we can use the metainfos to reconstruct the ports on the fly (and for all cells & subcells)

http://sami.klayout.org/preview?url=git%2Bhttps%3A%2F%2Fgithub.com%2Fgdsfactory%2Fmetainfo-ports.git%2Bklayout%5Bv0.0.6%5D

Should make a new button in klayout named "Show Ports" (and after clicking it "Hide Ports")

In the dropdown of the button all or specific cells can also be enabled/disabled.

This package should also be part of the gdsfactory klayout package I think (if I understand it correctly, but I am not sure it has been updated yet)

johnyb115, 2024-06-12:
Thanks a lot, works great. Now I see it's also mentioned in the updated tutorial for Klayout integration, as a separate package.

ZhimingShi, 2024-06-13:
> why not use klayout?
> 
> ```
> c = gf.components.straight()
> c.show()
> ```
> 
> the current plot has a scale bar ![Image](https://private-user-images.githubusercontent.com/4514346/338650423-aae15b7b-5788-450f-a07d-803333d0ce96.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgyNDM5NDksIm5iZiI6MTcxODI0MzY0OSwicGF0aCI6Ii80NTE0MzQ2LzMzODY1MDQyMy1hYWUxNWI3Yi01Nzg4LTQ1MGYtYTA3ZC04MDMzMzNkMGNlOTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTNUMDE1NDA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2IyNjcxMzQ4MGQ0NjA2YmFkMzYwZGI2OTg1ODRiNWNmNzhjYjQ2OTRkOGEwYjhlNWNjYTgxN2U5NjE4NTM3MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.hy7O4pVAjThc9q30-KDYXViCAK_sqbP8GwX7gy-WoT0)
> 
> my main issue with c.plot_matplotlib() was that it was always too slow for complex layouts
> 
> We also have a widget available but i never use it as i find it too slow
> 
> ![Image](https://private-user-images.githubusercontent.com/4514346/338655525-23e0e435-6833-49bf-afcd-7a149f6dee93.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgyNDM5NDksIm5iZiI6MTcxODI0MzY0OSwicGF0aCI6Ii80NTE0MzQ2LzMzODY1NTUyNS0yM2UwZTQzNS02ODMzLTQ5YmYtYWZjZC03YTE0OWY2ZGVlOTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTNUMDE1NDA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjEwMjI3NGFkY2RiZjgzYTIzNTA3ZjkyMmQ4M2RkOWJkMmJhZDA4ODcyMTkwZTgwMGEwNzA5YjRkZjJlN2E0ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.VnmCk5tHCuvEBH51rHrbOjEZ5kLK2Z9z3eusTWcgMdE)

Hello！c.plot_matplotlib() is really slow when looking at complex shapes, but I generally don't use it to display complex shapes, only to look at the submodules that make up complex shapes. This is because it has a coordinate system. klayout has no obvious coordinate system. Still wish I could add this plotting function.

johnyb115, 2024-06-13:
What do you mean? KLayout has great coordinate system, it can show grid, has different types of rulers for you to use etc.

sebastian-goeldi, 2024-06-13:
I am not sure I follow, KLayout should give you all the tools you need to find out certain coordinates. The annotation tools are probably what you are looking for

[image]

simbilod, 2024-07-08:
@ZhimingShi if it's any help, you can always extract the polygons from the Component yourself and plot them using Matplotlib. I have found this marginally useful when automating multiple plots (e.g. multiple colocated layouts + simulation results).

Here is an example:

```
import gdsfactory as gf
import matplotlib.pyplot as plt
from matplotlib.patches import Polygon

# Some Component
c = gf.components.bend_euler(cross_section="rib")

dbu = 1000 # 1 nm = 1 dbu, but Component is in microns

# Get the polygons
polygons_dict = c.get_polygons()

# Mapping between layer number and color
colors = {1: {"facecolor": "red", "alpha": 0.5},
            5: {"facecolor": "blue", "alpha": 0.5},
            }

# Plot the polygons
fig,ax = plt.subplots()

for layer_number, polygons in polygons_dict.items():

    color_dict = colors[layer_number]

    for polygon in polygons:
        p = Polygon([(point.x / dbu, point.y / dbu) for point in polygon.each_point_hull()], **color_dict)
        ax.add_patch(p)

# Set the bounds according to the Component bbox
bounds = c.bbox_np()
ax.set_xlim([bounds[0][0] - 2,bounds[1][0] + 2])
ax.set_ylim([bounds[0][1] - 2,bounds[1][1] + 2])

# Show
plt.xlabel("x (um)")
plt.ylabel("y (um)")
plt.gca().set_aspect('equal')
plt.show()
```

This yields:

[image]

And you should be able to control everything about the plot using Matplotlib.

closed 2024-07-08 by simbilod