gdsfactory/gdsfactory #2857: What motivated the change to `gf.components.bbox`?

95-martin-orion, 2024-06-11:
Prior to v8.0.0, the [`gf.components.bbox`](https://github.com/gdsfactory/gdsfactory/blob/351c213748e7fda0a46e05a945202e0a6b79e697/gdsfactory/components/bbox.py#L38) function created a simple rectangle on the given layer with the given lower left and upper right coordinates. I used this often as a shorter alternative to `gf.Component.add_polygon(...)` or `gf.components.rectangle(...).move(...)`.

After v8.0.0, the same function now takes a separate component and returns the bbox of that component plus some offset. What motivated this change? Can I expect the original `bbox` to be made available again, or should I switch to `add_polygon` for the old behavior?

sebastian-goeldi, 2024-06-12:
I am not entirely sure why the old behavior was completely dropped, but the motivation is to stop using cells where cells aren't really necessary and add complexity to the hierarchy. The bbox is a prime example. From a logic perspective it usually makes no sense to have the box be its own cell (unless you really want bc for example you want it to have ports).

As for how to do it, I have an offer that might not be too bad:

`box = gf.kdb.DBox(left=xmin, bottom=ymin, right=xmax, top=ymax # rectangle from xmin to xmax and height from ymin to ymax`

There are also two (3 if you count the point constructor) convenience constructors:
`gf.kdb.DBox(w=width, h=height) # create rectangle with fron -width/2 to wifth/2 and -height/2 to height/2`
`gf.kdb.DBox(w=width) # square centered at 0,0`
(`gf.kdb.DBox(lower_left=lower_left_dpoint, upper_right=upper_right_dpoint)`)

The returened boxes can be moved with `.move(dx=x,dy=y)`.

To finally insert it, use `c.shapes(my_layer_index).insert(box)`

Alternatively you can also assign left, bottom, right, top afterwards with `box.left = ...` (xmin, ymin, xmax, ymax)

Note:

- This inserts the box directly into the compnent, i.e. the component is still flat after this
- Do not rotate a box unless you need it as a bounding box (if you rotate it through `.transformed(gf.kdb.DCplxTrans(mag=1, rot=30, mirrx=False, x=0, y=0)` for example, you will just get a bigger box back as  a box cannot be off-grid; to do so you need to first cast it to a polygon with `polybox = gf.kdb.DPolygon(box)`

joamatab, 2024-06-12:
Hi Orion, 

the old behavior of bbox, is supported as `Component.bbox_np()` 

you can see the migration [guide](https://gdsfactory.github.io/gdsfactory/notebooks/21_migration_guide_7_8.html)

95-martin-orion, 2024-06-12:
> the old behavior of bbox, is supported as `Component.bbox_np()`

`Component.bbox_np` replaces `Component.bbox`, which returns the bbox of an existing component. I'm asking about `gf.components.bbox`, which _creates_ an axis-aligned rectangle.

> the motivation is to stop using cells where cells aren't really necessary and add complexity to the hierarchy. The bbox is a prime example.

That's a fair point. I think for my use case `comp.add_polygon(gf.kdb.DBox(...), layer)` might be more convenient, since we mostly operate with (layer, datatype) tuples rather than the layer indices that `shapes` takes, but the new LayerMap behavior might change that.

Thanks!

closed 2024-06-12 by 95-martin-orion