gdsfactory/gdsfactory #2859: add_fiber_array reports duplicate port names

MarcSchneiderD, 2024-06-12:
There are examples for "gdsfactory.routing.add_fiber_array", "gdsfactory.routing.add_pads_bot", "gdsfactory.routing.add_pads_top" in tutorial "04_routing_electrical.ipynb". These examples all show the error "ValueError: Found duplicate port names: o2: ..." or "ValueError: Found duplicate port names: e1: ..."
If you print the components ports with print(c.ports), only ports with individual names without duplicates are shown.

I could reproduce the bug with this simple code:
      c = gf.components.mzi()
      print(c.ports)
      cc = gf.routing.add_fiber_array(c)
      cc.plot()

The second line gives:
  ['Port(name: o1, dwidth: 0.5, trans: r180 *1 -10,0, layer: WG, port_type: optical)', 'Port(name: o2, dwidth: 0.5, trans: r0 *1 81.12,0, layer: WG, port_type: optical)']

The third line throws the error:
  ValueError: Found duplicate port names: o2: 2 If this intentional, please pass `check_ports=False` to the @cell decorator

Expected behavior: add an array of grating couplers or pads and routing from all ports to the couplers/pads.

I'm using Python 3.11.9 and gdsfactory 8.0.0. The bug didn't exist in gdsfactory 7.27.0.

joamatab, 2024-06-13:
yes, Marc,

it should be fixed in the main branch from github,

can you try
```
pip install git+https://github.com/gdsfactory/gdsfactory --force-reinstall
```

and see if it works as expected?

MarcSchneiderD, 2024-06-13:
With this it works. At least it doesn't throw error messages. Why is it different from pip install gdsfactory, which also leads to a version 8.0.0?
Well, add_fiber_array works now, even if routing is still a little weird...
[image]

joamatab, 2024-06-13:
We can release a new patch with the latest fixes

joamatab, 2024-06-13:
for fixing the issue you can use the route_south function first

MarcSchneiderD, 2024-06-13:
For that, I'll have to get more familiar with gdsfactory. Thank you!

closed 2024-06-13 by MarcSchneiderD