gdsfactory/gdsfactory #2861: v8.0.0: ComponentAlongPath is not hashable

95-martin-orion, 2024-06-12:
**Describe the bug**
As of v8.0.0, attempting to use [`gf.cross_section.ComponentAlongPath`] in `route_single_from_steps` raises the error:
```
TypeError: unhashable type: 'ComponentAlongPath'
```

**To Reproduce**
Create a `CrossSection` with `components_along_path` and use it as the `cross_section` in `route_single_from_steps`.

**Expected behavior**
The wire should be routed appropriately with the relevant component added along its path.

**Suggested fix**
This seems like a breakdown in the hashability of `CrossSection` - ensuring that `ComponentsAlongPath` is hashable should resolve this.

joamatab, 2024-06-14:
Hi Orion,

can you pass a tuple instead?

```
(gf.cross_section.ComponentAlongPath,)
```

95-martin-orion, 2024-06-14:
Using a tuple returns the same error.

<details><summary>Sample code</summary>

```python
# config parameters
wire_layer_tuple = (1, 0)
wire_width = 1
wire_radius = 10

wire_section = gf.Section(
    layer=wire_layer_tuple,
    width=wire_width,
    port_names=("start", "end"),
    port_types=("electrical", "electrical"),
)
wire_cross_section = gf.CrossSection(
    sections=[wire_section], radius=wire_radius
)

waypoints = [(0, 0), (100, 0), (100, 50), (20, 50), (20, 20)]
xover_layer_tuple = (2, 0)

@gf.cell
def bar() -> gf.Component:
    comp = gf.Component()
    rect = gf.components.rectangle(
        size=(wire_width / 2, 2 * wire_width),
        centered=True,
        layer=xover_layer_tuple,
    )
    comp.add_ref(rect)
    return comp

xovers = gf.cross_section.ComponentAlongPath(
    component=bar(), spacing=10, padding=5,
)
xover_cross_section = gf.CrossSection(
    sections=[wire_section],
    radius=wire_radius,
    components_along_path=(xovers,),
)

@gf.cell
def endpoint() -> gf.Component:
    comp = gf.Component()
    comp.add_port(
        "port",
        center=(0, 0),
        orientation=0,
        cross_section=xover_cross_section,
        port_type="electrical",
    )
    return comp

@gf.cell
def foo() -> gf.Component:
    comp = gf.Component()
    start = comp.add_ref(endpoint())
    end = comp.add_ref(endpoint())
    end.drotate(90).dmove(waypoints[-1])

    route = gf.routing.route_single_from_steps(
        comp,
        start.ports["port"],
        end.ports["port"],
        steps=[{"x": wp[0], "y": wp[1]} for wp in waypoints[1:-1]],
        bend=gf.components.bend_circular,
        cross_section=xover_cross_section,
        port_type="electrical",
    )
    return comp

my_foo = foo()
my_foo.plot()
```

</details>

The trace encounters the error while trying to retrieve a `bend_circular` invocation from the cache, which originates from this call:
```python
gf.get_component(
    component=gf.components.bend_circular,
    cross_section=xover_cross_section,
    radius=10.0,
)
```

95-martin-orion, 2024-06-21:
Update: I was able to circumvent this issue with the following hack:
```python
import attrs

# make a hashable equivalent of ComponentAlongPath
@attrs.frozen
class CAP:
    component: gf.Component
    spacing: float
    padding: float = 0.0
    offset: float = 0.0

# In the sample code above...
xovers = CAP(component=bar(), spacing=10, padding=5)
xover_cross_section = gf.CrossSection(
    sections=(wire_section,), # important: must be tuple, not list
    radius=wire_radius,
    components_along_path=(xovers,),
)
```
Note that changing `sections` to a tuple isn't enough - both it and `ComponentAlongPath` must be changed.

closed 2024-07-03 by joamatab