gdsfactory/gdsfactory #2862: Migration script over-capturing

fp-matt, 2024-06-12:
**Describe the bug**
The current migration script over-captures and replaces any parameter name that mirrors a deprecated attribute names

**To Reproduce**
For example, just ran the script on one of my files and found this:
[image]

**Expected behavior**
Only replace occurrences of deprecated attributes and not replace function parameters that mirror deprecated attribute names

**Suggested fix**
Maybe the current regex conditions are too loose? From what I see in `cli.py`, it is matching against any occurrence between word boundaries. Are there any occurrences you guys know of that would need to be replaced that aren't explicitly class attributes?

If not, I would suggest using a positive lookbehind to find a period instead of using the first word boundary so that a) the period isn't included in the capturing group, and b) it more tightly restricts the match criteria

for example:
```
 to_be_replaced = {
        "center",
        "mirror",
        "move",
        "movex",
        "movey",
        "rotate",
        "size_info",
        "x",
        "xmin",
        "xmax",
        "xsize",
        "y",
        "ymin",
        "ymax",
        "ysize",
    }

[...]

 pattern1 = re.compile(
        r"(?<=\.)(" + "|".join(r"d\." + _r for _r in to_be_replaced) + r")\b"
    )
    pattern2 = re.compile(r"(?<=\.)(" + "|".join(to_be_replaced) + r")\b")
    replacement = r"d\1"
```

Of course, a different solution would have to be found in the case that there are substitutions that need to be made that aren't class attributes

sebastian-goeldi, 2024-06-13:
Yes, I am aware that it's pretty loosely defined regexes. The problem is that certain constructors also needed this change (but I am not 100% sure that applies to gdsfactory7 code). I am okay adding the lookbehind behavior for dot as a new regex and make it the default behavior but allow something like `--class-attrs-only` (default if no flag) vs `--all` or something like that.

Would that be a satisfactory proposition?

(cc @joamatab as this would also necessitate some modification to the migration guide to document the new behavior)

closed 2024-06-16 by joamatab

fp-matt, 2024-06-17:
I know the PR for this has already been merged, but for what it's worth, I think that sounds great!