gdsfactory/gdsfactory #2864: add_pins not working properly

duarte-jfs, 2024-06-13:
I have found three issues. two I believe to have solved, the other I don't know how.

On the `gf.add_pins.add_pins()` function, The arguments that are passed to `function` are now:

```python
for port in ports:
        function(component=component, port=port, **kwargs)
```

However, in kwargs, the `layer` is not included. Therefore, this line must be corrected to:

```python
for port in ports:
        function(component=component, port=port, layer = layer, **kwargs)
```

Another one is in `gf.component.add_label()`. In it, the `position` is checked if it is an instance of tuple or list. However, when using `gf.add_pins_triangle()` it passes an argument to `gf.component.add_label()` where `position` is an ndarray, and raises an error. I propose the following change:

```python
elif isinstance(position, tuple | list ):
            x, y = position
```

to 

```python 
elif isinstance(position, Iterable ):
            x, y = position
```

The third issue is when using the `gf.add_pins_triangle()`. This creates huge triangles (see below). I don't know what is the best way to solve it.

[image]

duarte-jfs, 2024-06-13:
I also propose that the `add_pins()` is changed:

From:
```python
def add_pins(
    component: Component,
    port_type: str | None = None,
    layer: tuple[int, int] | None = None,
    orientation: int | None = None,
    width: float | None = None,
    function: Callable = add_pin_rectangle_inside,
    **kwargs,
) -> None:
    """Add Pin port markers.

    Args:
        component: to add ports to.
        port_type: optical, electrical, ...
        layer: layer for the pin marker.
        orientation: orientation for the pin marker.
        width: width for the pin marker.
        function: to add each pin.

    Keyword Args:
        kwargs: add pins function settings.
        pin_length: length of the pin marker for the port.
        layer: layer for the pin marker.
        layer_label: add label for the pin marker.
    """
    from gdsfactory.pdk import get_component

    component = get_component(component)

    #This should only select ports according to the port type
    ports = select_ports(
        ports=component.ports,
        port_type=port_type,
        layer=layer,
        orientation=orientation,
        width=width,
    )

    
    ports = component.ports

    for port in ports:
        function(component=component, port=port, **kwargs)
```

To:
```python
def add_pins(
    component: Component,
    port_type: str | None = None,
    layer: tuple[int, int] | None = None,
    orientation: int | None = None,
    width: float | None = None,
    function: Callable = add_pin_rectangle_inside,
    **kwargs,
) -> None:
    """Add Pin port markers.

    Args:
        component: to add ports to.
        port_type: Which port type do you want to add pins to. optical, electrical, ...  If None, it will add to all.
        layer: layer for the pin marker.
        orientation: orientation for the pin marker.
        width: width for the pin marker.
        function: to add each pin.

    Keyword Args:
        kwargs: add pins function settings.
        pin_length: length of the pin marker for the port.
        layer: layer for the pin marker.
        layer_label: add label for the pin marker.
    """
    from gdsfactory.pdk import get_component

    component = get_component(component)

    #This should only select ports according to the port type
    ports = select_ports(
        ports=component.ports,
        port_type=port_type,
    )

    for port in ports:
        function(component=component, port=port, layer = layer, **kwargs)
```

joamatab, 2024-06-14:
thank you for the suggested fixes

see https://github.com/gdsfactory/gdsfactory/pull/2868 for the fixes. I also added a test for this function

closed 2024-06-14 by joamatab