gdsfactory/gdsfactory #2866: nets as dicts

tvt173, 2024-06-14:
This is a continuation of https://github.com/gdsfactory/gdsfactory/issues/2831

We recently changed the netlist extraction to extract a list of "links" rather than a dictionary of "connections" for reasons mentioned in #2831. This proposal is to
- change the name of the "links" section to "nets"
- change the format from a list of lists to a list of dicts

## Justification
### changing the section name from "links" to "nets"
- "links" is already a subsection name within "routes" with a different format. it may be confusing to have two different sections with the same name but different formats
- "nets" is more of an industry-standard name, at least for electronics

### changing the format from a list of lists to a list of dicts
- this way we can add additional information to the nets, such as names, extracted lengths and attributes, etc.
- we may lose some compactness in exchange for simplicity. but since this section is intended to primarily be extracted or generated rather than hand-written, this is an acceptable exchange

@flaport 
@sebastian-goeldi 
@joamatab

sebastian-goeldi, 2024-06-14:
As for string vs list representation, Can we just check whether they are a mapping vs a list and build backwards compatibility (on read only) or does that not make sense?

tvt173, 2024-06-14:
> As for string vs list representation, Can we just check whether they are a mapping vs a list and build backwards compatibility (on read only) or does that not make sense?

You mean to support both the list-of-lists format we started supporting for "links" a few commits ago and this new list-of-dicts format? I think it would be easier to start simple. and i think dicts are more flexible. but i'm open for debate on what the entries in the dict should be.

i started with `p1` and `p2` with port strings specified in the same way we support elsewhere in the yaml format. I think we should also support `name` as an optional attribute. it may also be interesting to let the port definition be a dict like `{"inst": "my_inst1", "port": "o1"}` so that no string parsing is necessary

closed 2024-06-14 by joamatab

sebastian-goeldi, 2024-06-14:
I see. I am ok dropping old behavior. I was thinking maybe compatibility with old format makes sense. But since the category name changed, probably not

tvt173, 2024-06-14:
the "old" behavior of links was only around for a few days just now anyways. before that we used "connections" which was a big dictionary. we still parse connections on read, but they have a layout implication also. that's why we're moving to "nets" on extraction, so that they are purely for connectivity, and placement information is purely captured in "placements"

PR #2867 (2024-06-14): changing links to nets
fixes #2866