gdsfactory/gdsfactory #2869: Gdsfactory 8: cell.get_polygons() does not return datatypes

flaport, 2024-06-14:
**Describe the bug**
In gdsfactory<8 cell.get_polygons() returned (layer, datatype) tuples mapping to polygons. The current cell.get_polygons() only returns a layer index mapping to polygons. This is inconsistent with the previous version and moreover there is no way to retrieve the datatype from the polygons either.

**To Reproduce**
Try cell.get_polygons() and see for yourself.

**Expected behavior**
We want (layer, datatype) tuples as keys of the resulting dictionary

**Suggested fix**
Not sure. Haven't been able to figure it out.

sebastian-goeldi, 2024-06-14:
Can you inspect the exact type of the returned object?

To my knowledge it should be a `LayerEnum`, which has the desired info either with:

`layerindex.layer, layerindex.datatype`

Or you can directly cast it to a tuple for similar behavior to before

`tuple(layerindex)`

flaport, 2024-06-15:
No it's just an integer. Currently I'm using the integer to get the layer tuple from cell.layout().layer_infos(), but it's not super convenient.

PR #2876 (2024-06-15): fix #2869 by adding a 'by' argument to get_polygons and get_polygons_points
fix #2869 by adding a 'by' argument to get_polygons and get_polygons_points

@joamatab 
@sebastian-goeldi
commit 0bedfb0c (2024-06-15): fix #2869 by adding a 'by' argument to get_polygons and get_polygons_points
closed 2024-06-15 by joamatab