gdsfactory/gdsfactory #2881: Regression when creating an mzi. Maybe something wrong with reference.mirror() ?

flaport, 2024-06-16:
The following code works for both gdsfactory7 and 8:

```python
@gf.cell
def simple_mzi():
    c = gf.Component()

    # components
    mmi_in = gf.components.mmi1x2()
    mmi_out = gf.components.mmi2x2()
    bend = gf.components.bend_euler()
    half_delay_straight = gf.components.straight(length=10.0)

    # references
    mmi_in_ref = c << mmi_in
    mmi_out_ref = c << mmi_out
    straight_top1_ref = c << half_delay_straight
    straight_top2_ref = c << half_delay_straight
    bend_top1_ref = c << bend
    bend_top2_ref = (c << bend).mirror()
    bend_top3_ref = (c << bend).mirror()
    bend_top4_ref = c << bend
    bend_btm1_ref = (c << bend).mirror()
    bend_btm2_ref = c << bend
    bend_btm3_ref = c << bend
    bend_btm4_ref = (c << bend).mirror()

    # connections
    bend_top1_ref.connect("o1", mmi_in_ref.ports["o2"])
    straight_top2_ref.connect("o1", bend_top1_ref.ports["o2"])
    bend_top2_ref.connect("o1", straight_top1_ref.ports["o2"])
    bend_top3_ref.connect("o1", bend_top2_ref.ports["o2"])
    straight_top2_ref.connect("o1", bend_top3_ref.ports["o2"])
    bend_top4_ref.connect("o1", straight_top2_ref.ports["o2"])

    bend_btm1_ref.connect("o1", mmi_in_ref.ports["o3"])
    bend_btm2_ref.connect("o1", bend_btm1_ref.ports["o2"])
    bend_btm3_ref.connect("o1", bend_btm2_ref.ports["o2"])
    bend_btm4_ref.connect("o1", bend_btm3_ref.ports["o2"])

    mmi_out_ref.connect("o1", bend_btm4_ref.ports["o2"])

    # ports
    c.add_port(
        "o1",
        port=mmi_in_ref.ports["o1"],
    )
    c.add_port("o2", port=mmi_out_ref.ports["o3"])
    c.add_port("o3", port=mmi_out_ref.ports["o4"])
    return c
```

But both give pretty different results:

gdsfactory7 (as expected):
[image]

gdsfactory8 (unfortunately):
[image]

Not 100% sure, but the .mirror method of the references seem inconsistent.

sebastian-goeldi, 2024-06-17:
You have a small typo:

```python
# connections
bend_top1_ref.connect("o1", mmi_in_ref.ports["o2"])
straight_top1_ref.connect("o1", bend_top1_ref.ports["o2"]) # was straight_top2 but should be 1
```

and yes, there is a problem in kfactory. The initial port was taking over the mirror flag

flaport, 2024-06-17:
Thanks for the quick fix! Just checked (and fixed the typo locally) and it works!

flaport, 2024-06-17:
@joamatab can you make a GDS factory patch release depending on the fixed kfactory?

sebastian-goeldi, 2024-06-17:
I didn't release yet. I want to fix some routing issues first.

flaport, 2024-06-17:
Yeah, I think that there are also still some issues in gdsfactory that need to be fixed.

For example:

[image]

@joamatab 

opening this issue again for now.

sebastian-goeldi, 2024-06-29:
@flaport I think this is fixed now? I am not 100% sure, but yesterday @joamatab deployed several fixes

joamatab, 2024-07-01:
```python
import gdsfactory as gf 

@gf.cell
def simple_mzi():
    c = gf.Component()

    # components
    mmi_in = gf.components.mmi1x2()
    mmi_out = gf.components.mmi2x2()
    bend = gf.components.bend_euler()
    half_delay_straight = gf.components.straight(length=10.0)

    # references
    mmi_in_ref = c << mmi_in
    mmi_out_ref = c << mmi_out
    straight_top1_ref = c << half_delay_straight
    straight_top2_ref = c << half_delay_straight
    bend_top1_ref = c << bend
    bend_top2_ref = (c << bend).mirror()
    bend_top3_ref = (c << bend).mirror()
    bend_top4_ref = c << bend
    bend_btm1_ref = (c << bend).mirror()
    bend_btm2_ref = c << bend
    bend_btm3_ref = c << bend
    bend_btm4_ref = (c << bend).mirror()

    # connections
    bend_top1_ref.connect("o1", mmi_in_ref.ports["o2"])
    straight_top1_ref.connect("o1", bend_top1_ref.ports["o2"]) # was straight_top2 but should be 1

    straight_top2_ref.connect("o1", bend_top1_ref.ports["o2"])
    bend_top2_ref.connect("o1", straight_top1_ref.ports["o2"])
    bend_top3_ref.connect("o1", bend_top2_ref.ports["o2"])
    straight_top2_ref.connect("o1", bend_top3_ref.ports["o2"])
    bend_top4_ref.connect("o1", straight_top2_ref.ports["o2"])

    bend_btm1_ref.connect("o1", mmi_in_ref.ports["o3"])
    bend_btm2_ref.connect("o1", bend_btm1_ref.ports["o2"])
    bend_btm3_ref.connect("o1", bend_btm2_ref.ports["o2"])
    bend_btm4_ref.connect("o1", bend_btm3_ref.ports["o2"])

    mmi_out_ref.connect("o1", bend_btm4_ref.ports["o2"])

    # ports
    c.add_port(
        "o1",
        port=mmi_in_ref.ports["o1"],
    )
    c.add_port("o2", port=mmi_out_ref.ports["o3"])
    c.add_port("o3", port=mmi_out_ref.ports["o4"])
    return c

c = simple_mzi()
c.show()

```

this works great for me

flaport, 2024-07-02:
Hi @joamatab @sebastian-goeldi , thanks! I think we're getting really close!

Unfortunately, it's not yet fixed in `gf.read.from_yaml`. More specifically, the mirror argument is ignored in placements when a connection is made (see minimal example)... i.e. the resulting GDS looks the same when the mirror placement is there or not.

**minimal example:**

```python
import gdsfactory as gf
yaml = """
pdk: cspdk.si220
instances:
  c1:
    component: coupler_sc
    settings: {}
  b1:
    component: bend_sc
    settings: {}
  b2:
    component: bend_sc
    settings: {}
connections:
  c1,o3: b1,o1
  b2,o1: b1,o2
ports: {}
placements:
  b2:
    mirror: true
"""
c = gf.read.from_yaml(yaml)
c.show()
```

This is how the GDS looks:

[image]

In the case of `mirror: true` for bend `b2` the bend `b2` should point the other way.

joamatab, 2024-07-03:
this is how it looks with the fix

[image]

flaport, 2024-07-03:
@joamatab , still not correct. It should look like this:

[image]

flaport, 2024-07-03:
I suggest to use my fix in #2936 in stead.