gdsfactory/gdsfactory #2885: straight_all_angle is not working

duarte-jfs, 2024-06-17:
The `gdsfactory.components.straight_all_angle` cannot be placed on a component. It returns the error:

```python
AttributeError: 'ComponentAllAngle' object has no attribute 'layout'
```

The following snippet should reproduce the error:

```python
import gdsfactory as gf
from gdsfactory.components import straight_all_angle

c = gf.Component()

_a = straight_all_angle(length = 1000)
```

sebastian-goeldi, 2024-06-17:
```python
a = c << _a
```

This is your crucial difference. `<<` is equivalent to `c.create_inst(_a)`.

Since the upgrade to v8, all angle components (and instances) need to be treated differently.

AllAngle components __must__ be instantiated with `c.create_vinst(_a)`.

I'll see whether we can make a better error for this. 

For the record this error comes from when the cell is trying to check whether it and the to be instantiated cell belong to the same layout object. This fails because the virtual/all-angle components do not have such a strong tie to a layout object as the manhattan components.

duarte-jfs, 2024-06-18:
> ```python
> a = c << _a
> ```
> 
> This is your crucial difference. `<<` is equivalent to `c.create_inst(_a)`.
> 
> Since the upgrade to v8, all angle components (and instances) need to be treated differently.
> 
> AllAngle components **must** be instantiated with `c.create_vinst(_a)` (for Components, in ComponentAllAngle it can be different, but `create_vinst` is available in both).
> 
> I'll see whether we can make a better error for this.
> 
> For the record this error comes from when the cell is trying to check whether it and the to be instantiated cell belong to the same layout object. This fails because the virtual/all-angle components do not have such a strong tie to a layout object as the manhattan components.

Thank you so much for the reply! What would you say is the best way to solve it? I see that `ComponentAllAngle` is a child of `ComponentBase` which has the `__lshift__` defined for `ComponentReference(kf.KCell.create_inst(self, component))`. However, I am quite lost how the kfactory works, and I see that `ComponentAllAngle` also inherits from `kf.VKCell`... Can you give some pointers?

duarte-jfs, 2024-06-22:
For now I have made the following fix:

```python
class Component(ComponentBase, kf.KCell):
    """Canvas where you add polygons, instances and ports.

    - stores settings that you use to build the component
    - stores info that you want to use
    - can return ports by type (optical, electrical ...)
    - can return netlist for circuit simulation
    - can write to GDS, OASIS
    - can show in KLayout, matplotlib or 3D

    Properties:
        info: dictionary that includes derived properties, simulation_settings, settings (test_protocol, docs, ...)
    """

    def __init__(
        self,
        name: str | None = None,
        kcl: kf.KCLayout | None = None,
        kdb_cell: kdb.Cell | None = None,
        ports: kf.Ports | None = None,
    ):
        """Initializes a Component."""
        self.insts = ComponentReferences()
        super().__init__(name=name, kcl=kcl, kdb_cell=kdb_cell, ports=ports)

    def __lshift__(self, component: Component | ComponentAllAngle) -> ComponentReference:  # type: ignore[override]
        """Creates a Componenteference reference to a Component."""
        if isinstance(component, kf.KCell):
            return ComponentReference(kf.KCell.create_inst(self,component)) 
        
        elif isinstance(component, kf.VKCell):
            return kf.KCell.create_vinst(self, component)
```

This has allowed me to place `ComponentAllAngle` elements. However, like this we are not able to create a `ComponentReference` since those objects only accept `KInstance`. I'm not sure what we miss here.

sebastian-goeldi, 2024-06-23:
You miss that in gdsfactory8 instead of

```python
a = c << _a
```

You must use

```python
a = c.creat_vinst(_a)
```

No overload will save you, because it will latest break when you are trying to instantiate a real component as an all angle instance. This is why there is no overloaded `<<` for it. That was a deliberate choice, not an oversight.

duarte-jfs, 2024-06-23:
Ahhh I see! That means that in the future there is no plan to continue with the << operator? Will the "Component" and "reference" terminology, and consequently the ComponentReference, be discontinued?

sebastian-goeldi, 2024-06-23:
No, it will not.

There is now a split between "regular" (meaning manhattam -> 90° rotations and grid sizes only, therefor no 10.0005um long straights when dbu == 1nm) and "all angle" compon buents.

`<<` is reserved for the standard components in a component and will not work to instantiate an all amgle component.

On an all angle component `<<` works, because it only implements a subset of all available component functions and all instances are VInstances.

In a normal component we have much more options now such as recursive instance iterators and recursive shape iterators and other cool tools. But the architecture of klayout does not allow (easily) for non-manhattan components

sebastian-goeldi, 2024-06-23:
No, it will not.

There is now a split between "regular" (meaning manhattam -> 90° rotations and grid sizes only, therefore no 10.0005um long straights when dbu == 1nm) and "all angle" components.

`<<` is reserved for the standard components in a component and will not work to instantiate an all amgle component.

On an all angle component `<<` works (but I still would recommend `c.create_vinst` over it), because it only implements a subset of all available component functions and all instances are VInstances.

In a normal component we have much more options now such as recursive instance iterators and recursive shape iterators and other cool tools. But the architecture of klayout does not allow (easily) for non-manhattan components. Also booleans are more or less restricted to standard components.

duarte-jfs, 2024-07-01:
I see. I think I'm following what you're saying, but I still didn't get this:

> On an all angle component << works

I tried to do this on on an `bend_euler_all_angle` and it did not work. I have now switched to `.create_vinst()`. 

I also recognize this thread is no longer an "issue" but rather a discussion, so feel free to move over to the "Discussions" tab if you prefer. And thank you for your understanding and explanations!

sebastian-goeldi, 2024-07-01:
> I see. I think I'm following what you're saying, but I still didn't get this:

>>    On an all angle component << works

for that to work you need to have

```python
c = gf.ComponentAllAngle()
a = c << _a
```

duarte-jfs, 2024-07-01:
> > I see. I think I'm following what you're saying, but I still didn't get this:
> 
> > > On an all angle component << works
> 
> for that to work you need to have
> 
> ```python
> c = gf.ComponentAllAngle()
> a = c << _a
> ```

But running the snippet below:

```python
import gdsfactory as gf

c = gf.Component()
_a = gf.ComponentAllAngle()

a = c << _a
```

returns:

> AttributeError: 'ComponentAllAngle' object has no attribute 'layout'

I have gdsfactory 8.4.0 installed

sebastian-goeldi, 2024-07-01:
yes, because `c` NEEDS to be a `ComponentAllAngle` in which case `_a` CAN be a `Component` or `ComponentAllAngle`.
If `c` is a `Component`, `_a` MUST be a `Component`.

duarte-jfs, 2024-07-01:
ahh, yes, I missed that. That is my fault, sorry for the confusion. I see now that the `ComponentAllAngle` inherits the `__lshift__` from the `kf.VKCell` which wraps around the `.create_vinst()`. Everything makes sense now: as it stands, `ComponentAllAngle` and `Component` aren't meant to be combined in a single `Component`. I assume this is, as you said, for practical reasons and by design, not an oversight. From a user perspective though, it would be nice to not be aware of this distinction, but I understand how difficult it may be. Anyway, thank you for the help (and patience XD)!

sebastian-goeldi, 2024-07-01:
Yes, I am aware of that problem. I thought it would error out properly, but apparently it doesn't.

We are also continuously working on improving this, so you can expect that at some point this will create better error messages. I am sorry for the inconveniences. On the other hand we are fairly certain that this will cause less (hopefully this means 0) problems with snapping in the future

lodoreds, 2024-07-29:
It would be really useful to have this conversation integrated in the [grid snapping tutorial](https://gdsfactory.github.io/gdsfactory/notebooks/12_grid_snapping.html). At first, it was not clear to me that AllAngle components can only be instantiated with `create_vinst`.

clee5310, 2024-11-25:
Just like to second lodoreds' last comment.  
Also, I found component created by ComponentAllAngle cannot be written to a GDS file by write_gds() directly.  One must use "a = c.creat_vinst(_a)   # c is a Component" first before using c.write_gds(....) to generate a GDS file.