gdsfactory/gdsfactory #2887: gf.path.transition(width_type='linear') coming out with nonlinear width variation for some cases

jonnyroth, 2024-06-18:
**Describe the bug**
GDSFactory 7.23.0, Python 3.10.  Apologies if it's been corrected, I will upgrade to the latest version after my tapeout.
I'm using gf.path.transition, between two different gf.cross_section.rib_width_trenches.
Despite the two cross-sections having the same width=1, the actual width varies between ~0.58 and ~1.44.

**To Reproduce**
```python
import gdsfactory as gf

xs1 = gf.cross_section.rib_with_trenches(width=1,width_trench=5,layer=(1,0),layer_trench=(2,0))
xs2 = gf.cross_section.rib_with_trenches(width=1,width_trench=1,layer=(1,0),layer_trench=(2,0))
xst = gf.path.transition(xs1,xs2,width_type='linear')
xs1b = gf.cross_section.CrossSection(sections=[gf.cross_section.Section(width=5, layer=(1,0))])
xs2b = gf.cross_section.CrossSection(sections=[gf.cross_section.Section(width=1, layer=(1,0))])
xstb = gf.path.transition(xs1b,xs2b,width_type='linear')
path1 = gf.path.straight(length=100)
path2 = gf.path.euler(angle=90,p=0.2,radius=50,use_eff=True)
path3 = gf.path.arc(angle=90,radius=50)
top = gf.Component()
e1 = top << gf.path.extrude(path1,cross_section=xst)
e2 = top << gf.path.extrude(path2,cross_section=xst)
e3 = top << gf.path.extrude(path3,cross_section=xst)
e4 = top << gf.path.extrude(path3,cross_section=xstb)
e2.move((0,-75))
e3.move((0,-150))
e4.move((0,-225))
top.show()
```
[image]
Image: From top to bottom: Top (straight) is fine, Second (Euler) and Third (Arc) have erroneous width variation, Fourth (Arc using single layer waveguide) is fine.

**Expected behavior**
Slab total width changes linearly. waveguide width changes linearly, which in this case is staying constant.

**Suggested fix**
Update to gf.cross_section module, assuming it's not already fixed in v8.

tvt173, 2024-06-18:
hi @jonnyroth ,

interesting... not sure why exactly this is, but I imagine it must have to do with the offset in the trench section. i never usually use this cross section. i would usually draw as shown below, and then derive the trench via a boolean. please see the modified sample script:

```python
import gdsfactory as gf

def my_rib_with_trenches(width=1, width_trench=5, slab_offset=0.3, layer=(1,0), layer_trench=(2,0), layer_slab=(3,0)):
    core_width = width
    trench_width_full = 2* width_trench + core_width
    slab_width_full = trench_width_full + 2 * slab_offset
    return gf.cross_section.CrossSection(sections=[
        gf.cross_section.Section(name="rib",width=core_width, layer=layer), 
        gf.cross_section.Section(name="trench",width=trench_width_full, layer=layer_trench),
        gf.cross_section.Section(name="slab", width=slab_width_full, layer=layer_slab)])

xs1 = my_rib_with_trenches(width=1,width_trench=5,layer=(1,0),layer_trench=(2,0))
xs2 = my_rib_with_trenches(width=1,width_trench=1,layer=(1,0),layer_trench=(2,0))
xst = gf.path.transition(xs1,xs2,width_type='linear')

path1 = gf.path.straight(length=100)
path2 = gf.path.euler(angle=90,p=0.2,radius=50,use_eff=True)
path3 = gf.path.arc(angle=90,radius=50)
top = gf.Component()
e1 = top << gf.path.extrude(path1,cross_section=xst)
e2 = top << gf.path.extrude(path2,cross_section=xst)
e3 = top << gf.path.extrude(path3,cross_section=xst)
e2.move((0,-75))
e3.move((0,-150))
top.show()
```

[image]

fp-matt, 2024-06-18:
I've run @jonnyroth 's sample code (with some minor tweaks to get it in a compatible state, listed at the end of this comment)  and gdsfactory v8.1 appears to have the same behavior

[image]

From what I can gather, this seems to be an issue with `Path._centerpoint_offset_curve(...)`

To try to help track down the issue, I ran some data analysis on some variables of interest within `Path.extrude_transition(...)`

[image]

A couple things of note:
- There are some deviations in the values of `dy` between the upper and lower trenches, though these deviations are much smaller than the deviations we see in the ridge width.
- The non-constant width issue is immediately present in the trench polygons (created from the points generated by `_centerpoint_offset_curve(...)`)
- Interestingly, the "Deviation from Ideal Ridge Width" plot indicates that the deviation very closely approximates a sin function, though I have not run the analysis under different conditions to make sure this isn't just coincidence)
  - I'm curious if anyone else has an idea as to why this might be the case - granted I haven't actually spent a whole lot of time digging into `_centerpoint_offset_curve(...)`, so it may very well be obvious once actually looked at, but I've spent way too much time drafting up this comment in the first place, so I'll leave that part to someone else.

This is actually my first time seeing this issue because I do basically the same thing that @tvt173 does. As he explained in his comment, just block out the general shape you want, draw out the trenches, and then do some boolean subtraction to get the final shape.

----

code used to generate the analysis plots (note that any variable with `_upper` or `_lower` in its name denotes that it is derived from the `trench_1` or `trench_2` sections, respectively):
```
fig, ax = plt.subplots(1, 3, figsize=(15,5))

ax[0].plot(width_upper / 2, label="Upper Trench")
ax[0].plot(width_lower / 2, label="Lower Trench")
ax[0].set_xlabel("Point Index")
ax[0].set_ylabel("'$dy$' (um)")
ax[0].title.set_text("Trench Width Offsets")
ax[0].legend()

ax[1].plot(upper_trench[:, 0], upper_trench[:, 1], label="Upper Trench")  # taken from 'points_poly'
ax[1].plot(lower_trench[:, 0], lower_trench[:, 1], label="Lower Trench")  # taken from 'points_poly'
ax[1].set_xlabel("X (um)")
ax[1].set_ylabel("Y (um)")
ax[1].title.set_text("Upper & Lower Trench Polygons")
ax[1].legend()

# First glace, this line doesn't seem right to me, but plotting the raw upper and lower trench data indicates that 'points1_upper'
# and 'points2_lower' correspond to the inner edges of the polygons
s_diff = np.sqrt((points1_upper[:, 0] - points2_lower[:, 0]) ** 2 + (points1_upper[:, 1] - points2_lower[:, 1]) ** 2)

ax[2].plot(s_norm := s_diff - 1, linewidth=2)
x_min, x_max = ax[2].get_xlim()
ax[2].plot([-100, 1000], [0, 0], 'k',ls='dotted', linewidth=1)
ax[2].set_xlabel("Point Index")
ax[2].set_ylabel("$\Delta$w (um)")
ax[2].set_xlim(x_min, x_max)
ax[2].set_yticks([s_norm.min(), 0, s_norm.max()])
ax[2].title.set_text("Deviation from Ideal Ridge Width")

fig.tight_layout()
fig.show()
```
----

Minor tweaks to sample code needed for gdsfactory v8.1 compatibility:

- Provide `wg_marking_layer` to `xs1` and `xs2`: `rib_with_trenches` type hints it as optional, but since `wg_marking_layer` is given as the main layer of the cross-section, it cannot be None in order to extrude the cross-section
- `xs1b` and `xs2b` both need to have a `name` parameter defined in the `Section` defined in their `CrossSection`s. The same name must be used for both

PR #3130 (2024-08-14): fix #2887: allow specification of offset_type other than 'sine' in Transition & fix extrude_transition()
``Fixes #2887, where changes in `Section` offsets along a transition are tapered by a sine transition by default, even when width is tapered e.g. linearly.
This causes width variations along a transition waveguides, where the core is defined by trenches.
An `offset_type `(same logic as `width_type`) is introduced to allow same taper shapes for offset shapes as well.

Additionally, `extrude_transition()` is modified to extrude an offset shape along the actual path instead of extruding the ce [... cut]
commit 56c716bb (2024-08-14): fix #2887: allow specification of offset_type other than 'sine' in Transition & make extrude_transition() extrude along actual path not offset path
commit bbaed029 (2024-08-14): Merge pull request #3130 from gronniger/gr/fix2887

fix #2887: allow specification of offset_type other than 'sine' in Transition & fix extrude_transition()
closed 2024-08-14 by joamatab