gdsfactory/gdsfactory #2888: ValueError: xx is not a valid LAYER

SkyLin1020, 2024-06-18:
**Describe the bug**
ValueError: 45 is not a valid LAYER
in version 8.0.0, the code below works. But inversion 8.1.0, it does not work. the layer (37,1) seems be trasfered to a number '45'.
in c.getpolygons()[layer], same thing happened. there is no keys as (37,1) but keys as 45.

**To Reproduce**
s1 = gf.Section(width=1, offset=-1/2, layer=(37,1),name='wg', port_names=("o1", "o2"))
X1 = gf.CrossSection(sections=[s1])
s2 = gf.Section(width=1, offset=-1/2, layer=(37,1),name='wg', port_names=("o1", "o2"))
X2 = gf.CrossSection(sections=[s2])
Xtrans = gf.path.transition(cross_section1=X1, cross_section2=X2, width_type="linear")
P3 = gf.path.straight(length=10)
coupler_up = gf.path.extrude_transition(P3, Xtrans)

joamatab, 2024-06-18:
Hi Theodore,

you need to define that layer in the LayerMap

I noticed that we get layer 45 when you use a layer not defined in the layermap

SkyLin1020, 2024-06-18:
Hi Joamatab,

  thanks for your reply.
  I change my code as below
`
class LayerMapFab(LayerMap):
    COR: Layer = (37, 1)

LAYER = LayerMapFab

s1 = gf.Section(width=1, offset=-1/2, layer=LAYER.COR,name='wg', port_names=("o1", "o2"))
X1 = gf.CrossSection(sections=[s1])
s2 = gf.Section(width=1, offset=-1/2, layer=LAYER.COR,name='wg', port_names=("o1", "o2"))
X2 = gf.CrossSection(sections=[s2])
Xtrans = gf.path.transition(cross_section1=X1, cross_section2=X2, width_type="linear")
P3 = gf.path.straight(length=10)
coupler_up = gf.path.extrude_transition(P3, Xtrans)

`

the problem is still not resolved. Do I wrongly use LayerMap? Can you give me some advices?

joamatab, 2024-06-18:
i think i fixed your issue

can you try
```
pip install git+https://github.com/gdsfactory/gdsfactory --force-reinstall
``` 

and see if it works now?

closed 2024-06-18 by joamatab

SkyLin1020, 2024-06-19:
Hi @joamatab ，

it works now.
Thank you for your help.