gdsfactory/gdsfactory #2902: installation on Python in KLayout

lukasc-ubc, 2024-06-22:
**Describe the bug**

I am trying to install gdsfactory in KLayout, and I get error below.  This is on KLayout HW-klayout-0.29.1-macOS-Monterey-1-qt5MP-RsysPhb311.  Perhaps one of the gdsfactory dependancies does not want to install.  There are many successful dependancy installations.  

Is it possible to have an installation config option that skips some of the dependancies that fail?  

Any idea how to fix it?

thank you

```
> import sys
> sys.version
'3.11.9 (main, Apr  2 2024, 16:31:10) [Clang 14.0.0 (clang-1400.0.29.202)]'
> from SiEPIC import install
> install.install('gdsfactory')
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Defaulting to user installation because normal site-packages is not writeable
WARNING: Skipping /Users/lukasc/Library/Python/3.11/lib/python/site-packages/numpy-1.26.2.dist-info due to invalid metadata entry 'name'
WARNING: Skipping /Users/lukasc/Library/Python/3.11/lib/python/site-packages/numpy-1.26.2.dist-info due to invalid metadata entry 'name'
Collecting gdsfactory
  Downloading gdsfactory-8.2.3-py3-none-any.whl.metadata (11 kB)
Collecting jinja2<4 (from gdsfactory)
  Downloading jinja2-3.1.4-py3-none-any.whl.metadata (2.6 kB)
Collecting loguru<1 (from gdsfactory)
  Using cached loguru-0.7.2-py3-none-any.whl.metadata (23 kB)
Requirement already satisfied: matplotlib<4 in /Users/lukasc/Library/Python/3.11/lib/python/site-packages (from gdsfactory) (3.9.0)
Requirement already satisfied: numpy<2 in /Users/lukasc/Library/Python/3.11/lib/python/site-packages (from gdsfactory) (1.26.4)
Collecting omegaconf<3 (from gdsfactory)
  Using cached omegaconf-2.3.0-py3-none-any.whl.metadata (3.9 kB)
Collecting orjson<4 (from gdsfactory)
  Downloading orjson-3.10.5-cp311-cp311-macosx_10_15_x86_64.macosx_11_0_arm64.macosx_10_15_universal2.whl.metadata (49 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 49.7/49.7 kB 1.4 MB/s eta 0:00:00
Requirement already satisfied: pandas<3 in /Users/lukasc/Library/Python/3.11/lib/python/site-packages (from gdsfactory) (2.2.2)
Collecting pydantic<2.8,>=2 (from gdsfactory)
  Downloading pydantic-2.7.4-py3-none-any.whl.metadata (109 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 109.4/109.4 kB 3.1 MB/s eta 0:00:00
Collecting pydantic-settings<3 (from gdsfactory)
  Downloading pydantic_settings-2.3.3-py3-none-any.whl.metadata (3.3 kB)
Collecting pydantic-extra-types<3 (from gdsfactory)
  Downloading pydantic_extra_types-2.8.2-py3-none-any.whl.metadata (3.2 kB)
Requirement already satisfied: pyyaml in /Users/lukasc/Library/Python/3.11/lib/python/site-packages (from gdsfactory) (6.0.1)
Collecting qrcode (from gdsfactory)
  Using cached qrcode-7.4.2-py3-none-any.whl.metadata (17 kB)
Collecting rectpack<1 (from gdsfactory)
  Using cached rectpack-0.2.2-py3-none-any.whl
Requirement already satisfied: rich<14 in /Users/lukasc/Library/Python/3.11/lib/python/site-packages (from gdsfactory) (13.7.1)
Requirement already satisfied: scipy<2 in /Users/lukasc/Library/Python/3.11/lib/python/site-packages (from gdsfactory) (1.13.1)
Collecting shapely<3 (from gdsfactory)
  Downloading shapely-2.0.4-cp311-cp311-macosx_10_9_x86_64.whl.metadata (7.0 kB)
Collecting toolz<1 (from gdsfactory)
  Using cached toolz-0.12.1-py3-none-any.whl.metadata (5.1 kB)
Collecting types-PyYAML (from gdsfactory)
  Using cached types_PyYAML-6.0.12.20240311-py3-none-any.whl.metadata (1.5 kB)
Collecting typer<1 (from gdsfactory)
  Downloading typer-0.12.3-py3-none-any.whl.metadata (15 kB)
Requirement already satisfied: watchdog<5 in /Users/lukasc/Library/Python/3.11/lib/python/site-packages (from gdsfactory) (4.0.1)
Collecting kfactory==0.17.3 (from kfactory[git,ipy]==0.17.3->gdsfactory)
  Downloading kfactory-0.17.3-py3-none-any.whl.metadata (4.5 kB)
Collecting freetype-py (from gdsfactory)
  Using cached freetype_py-2.4.0-py3-none-macosx_10_9_universal2.whl.metadata (6.3 kB)
Collecting mapbox_earcut (from gdsfactory)
  Downloading mapbox_earcut-1.0.1-cp311-cp311-macosx_10_9_x86_64.whl.metadata (2.0 kB)
Collecting networkx (from gdsfactory)
  Downloading networkx-3.3-py3-none-any.whl.metadata (5.1 kB)
Collecting scikit-image (from gdsfactory)
  Downloading scikit_image-0.24.0-cp311-cp311-macosx_10_9_x86_64.whl.metadata (14 kB)
Collecting trimesh<4.4,>=4 (from gdsfactory)
[... 298 lines cut ...]
ERROR: Could not find a version that satisfies the requirement pygdsfactory (from versions: none)
ERROR: No matching distribution found for pygdsfactory
WARNING: Skipping /Users/lukasc/Library/Python/3.11/lib/python/site-packages/numpy-1.26.2.dist-info due to invalid metadata entry 'name'
WARNING: Skipping /Users/lukasc/Library/Python/3.11/lib/python/site-packages/numpy-1.26.2.dist-info due to invalid metadata entry 'name'
WARNING: Skipping /Users/lukasc/Library/Python/3.11/lib/python/site-packages/numpy-1.26.2.dist-info due to invalid metadata entry 'name'
False
> import gdsfactory
ModuleNotFoundError: No module named 'gdsfactory'
  

```

**To Reproduce**

See above.

[image]

**Expected behavior**

that it installs and imports correctly

**Suggested fix**

Making the import work.

joamatab, 2024-06-22:
I've never tried installing it in the macro development from klayout,

we usually recommend using klayout as a viewer and installing gdsfactory on a separate python

lukasc-ubc, 2024-06-22:
Here is a second attempt that works:

```
> import pip
> pip.main(['install','gdsfactory'])
> import gdsfactory
```

And it ran successfully.  I was able to get the a layout example to run and save a layout.  

Now, on to trying to display that layout directly, without saving and loading.  Any ideas how @joamatab ?

@mustafacc could you please check if this works on Linux?

@seanlam97 could you please check if this works on Windows?

thank you

mustafacc, 2024-06-23:
on ubuntu 22.04 it's failing

```
> pip.main(['install', 'gdsfactory'])
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Defaulting to user installation because normal site-packages is not writeable
Collecting gdsfactory
  Using cached gdsfactory-8.2.3-py3-none-any.whl (636 kB)
Collecting loguru<1
  Using cached loguru-0.7.2-py3-none-any.whl (62 kB)
Collecting trimesh<4.4,>=4
  Using cached trimesh-4.3.2-py3-none-any.whl (693 kB)
Requirement already satisfied: numpy<2 in /home/mustafa/.local/lib/python3.10/site-packages (from gdsfactory) (1.26.3)
Collecting rectpack<1
  Using cached rectpack-0.2.2.tar.gz (17 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [2 lines of output]
      Info: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
      ERROR: Unknown option: --egg-base
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output. 

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
1
```

It seems the issue is with installing the package `rectpack` itself.

lukasc-ubc, 2024-06-23:
Could you try installing with the "--no-dependencies" option added?

```
> import pip
> pip.main(['install','--no-dependencies','gdsfactory'])
> import gdsfactory
```

Maybe this can skip the installation of rectpack, and we can see if it is necessary.

sebastian-goeldi, 2024-06-23:
I would highly suggest to move away from installing directly in klayout. It can completely brick systems if used carelessly, on MacOS it can brick your (ana)conda env (since it most likely uses the default brew env).

Debian (and most likely other linux distros) are moving towards disallowing people to just install system wide (or even user space) packages with no venv [PEP](https://peps.python.org/pep-0668/)

This is to safeguard users from accidentially bricking vital system packages like this https://forums.linuxmint.com/viewtopic.php?t=381919
https://forums.centos.org/viewtopic.php?t=74134

Sure, there are ways to re-enable the old behavior, but that would feel like teaching kids to play on the train tracks in front of the house because it's easily viewable from the balcony.

mustafacc, 2024-06-23:
@lukasc-ubc It installs okay without dependencies, however, import fails when attempting to load kfactory (another package thats failing to install).

@sebastian-goeldi Agreed. I ran this on a VM to not cause any issues with the system's python. I was investigating https://github.com/flaport/condalayout to potentially mitigate this issue on linux but I ran into [other issues](https://github.com/flaport/condalayout/issues/3).

sebastian-goeldi, 2024-06-24:
Btw it works (on linux, but macos should be the same, just different python) with the following which will not abuse/pollute the system/user env:

```bash
cd ~
mkdir kfactory_klayout && cd kfactory_klayout
python3 -m venv .env
source .env/bin/activate
KLAYOUT_PYTHONPATH=~/kfactory_klayout/.env/lib/python3.11/site-packages/ klayout
```

with this I was able to successfully import and after a quick test use kfactory.

But be advised, at least with the debug console open, it takes a good 10 seconds to import kfactory (not even talking about gdsfactory here).

mustafacc, 2024-07-05:
I tried it but I seem to be running into an issue importing encodings module, even though it can be imported within that env.

```
(.env) (base) mustafa@mustafa:~/kfactory_klayout$ klayout 
Info: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = '/proc/61953/exe'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/proc/61953/exe'
  sys.base_prefix = ''
  sys.base_exec_prefix = ''
  sys.platlibdir = 'lib'
  sys.executable = '/proc/61953/exe'
  sys.prefix = ''
  sys.exec_prefix = ''
  sys.path = [
    '/home/mustafa/kfactory_klayout/.env/lib/python3.12/site-packages/',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007d6d244c3840 (most recent call first):
  <no Python frame>

```
echo `$KLAYOUT_PYTHONPATH: /home/mustafa/kfactory_klayout/.env/lib/python3.12/site-packages/`
which python: `/home/mustafa/kfactory_klayout/.env/bin/python`
python --version: `Python 3.12.2`

sebastian-goeldi, 2024-07-05:
You need to use system python on linux. Or to be more accurate, the python klayout was compiled against. And that, unless you compiled it yourself is probably not 3.12.