gdsfactory/gdsfactory #2903: saved layout has two extra cells

lukasc-ubc, 2024-06-22:
**Describe the bug**

When you save the layout using KLayout in Python, there are two extra cells: straight, and taper.

**To Reproduce**
```
import gdsfactory as gf
import pya

c = (
    gf.Component()
)  

mmi = c.add_ref(gf.components.mmi1x2())

layout = c.layout()

print('%s' % layout.cells())

layout.write('/tmp/1.gds')
c.write_gds('/tmp/2.gds')

```            

[image]

[image]

joamatab, 2024-06-22:
Hi Lucas,

we usually don't mix pya, (the GUI API for klayout) and `gf.kdb` (the python API)

something klayout does internally to store the metadata from Component.ports, component.info and component.settings is to create a second cell

lukasc-ubc, 2024-06-22:
Hi Joaquin,

I was running this test in VSCode, not in the GUI.  I removed the line "import pya".  I still get the extra cells.  My code did not instantiate the straight and taper cells.  So there must be some memory leak or other bug in gdsfactory that is creating these additional cells, and reducing the performance.

joamatab, 2024-06-23:
hi Lukas,

this works great for me

```python
import gdsfactory as gf

c = gf.Component()
mmi = c.add_ref(gf.components.mmi1x2())
c.write_gds('/tmp/2.gds')
c.show()
```

lukasc-ubc, 2024-06-23:
Hi @joamatab 

Indeed, there are two ways to save the file:

```
layout.write('/tmp/1.gds')
c.write_gds('/tmp/2.gds')
```

The 2nd one works, as shown in the screenshots above.  The 1st method works but adds two extra cells.

sebastian-goeldi, 2024-06-23:
The 2 extra cells can come from double wrapped components

If you do

```python
@gf.cell
def my_straight():
    return gf.c.straight()
```

The `@cell` will detect that the initial cell is already locked and create a new one. This is most likely where the extra cells come from.

So indeed `c.write_gds` is the best approach as the layout/kcl write will all cells in the layout. You can selectively exclude within the save_options though

sebastian-goeldi, 2024-06-23:
Also, please be aware, if you resort to `layout().write(...)` etc, you are circumventing a lot of gdsfactory sanitory checks as well as dropping (most if not all) meta infos and therefore will not be able to use any operation relying on read and reconstructed meta info such as http://sami.klayout.org/preview?url=git%2Bhttps%3A%2F%2Fgithub.com%2Fgdsfactory%2Fmetainfo-ports.git%2Bklayout%5Bv0.0.6%5D or when reading gds files into gdsfactory again.