gdsfactory/gdsfactory #2905: add_ref will truncate long name s

SkyLin1020, 2024-06-23:
**Describe the bug**
when use add_ref, long names will be truncated. and if there are many cells have similar name, in klayout it will be add '#num'

**To Reproduce**
```
c1 = gf.Component('Convolutional_4stage_1270_1510_FSR150_V1_bias_0')
c2 = gf.Component('Convolutional_4stage_1270_1510_FSR150_V1_sub_20.0')

c = gf.Component('cell0')

c << c1
c << c2
c.write_gds('test.gds')
```

**Expected behavior**
in kalyout, c1 and c2 will be named as 'Convolutional_4stage_1270_1510_F' and 'Convolutional_4stage_1270_1510$1'. how to avoid this issues. I want their names the same as what I give them.

joamatab, 2024-06-23:
Hi, 

this works well for me,

```
import gdsfactory as gf

c = gf.Component('Convolutional_4stage_1270_1510_FSR150_V1_sub_20.0')
print(c.name)
```

are you sure you are using the latest gdsfactory version?

can you `pip install gdsfactory --upgrade` and try again?

sebastian-goeldi, 2024-06-23:
The names are shortened on write. That's why there's a difference between the initial runtime name and the actual name in klayout. The easiest way to avoid having the names shortened is to set `CONF.gds2_max_cellname_length` to a high enough number.

If you want to completely remove it, I think you can set it to 0 or if that doesn't work `None` if pydantic doesn't complain.

SkyLin1020, 2024-06-24:
Hi, @sebastian-goeldi 

thank you for your help. do you mean the code like below?

```
import gdsfactory as gf

gf.CONF.max_cellname_length = 1000

c1 = gf.Component('Convolutional_4stage_1270_1510_FSR150_V1_bias_0')
c2 = gf.Component('Convolutional_4stage_1270_1510_FSR150_V1_sub_20.0')

c = gf.Component('cell0')

c << c1
c << c2
c.write_gds('test.gds')
```
it won't work. 
gf.CONF.max_cellname_length = 0 or a large number does not work either. and None will report an error.

SkyLin1020, 2024-06-24:
Hi, @joamatab 

I'm sorry for not being clear in the issue. I think sebastian-goeldi understanding it correctly. when print the cell name in code, it is original one,. but when write to gds file and see it in klayout , it is shortened.
[image]

hope the issue is clearly stated this time and there is a method to avoid this.

sebastian-goeldi, 2024-06-24:
Hmm, interesting. Sorry for the confusion and making you explain it twice. The last example definitely should work. I'll try to find out whether it is a gdsfactory bug or whether it is deeper down in kfactory. This is definitely not the intended behavior.

tvt173, 2024-06-24:
indeed this was a bug in the `write_gds` function. see my pending PR https://github.com/gdsfactory/gdsfactory/pull/2909. thanks @SkyLin1020 for reporting it and @sebastian-goeldi for helping track it down!

PR #2909 (2024-06-25): save_options should be None by default
Currently the default value of `save_options` in `write_gds` is fetched from the config at the time of declaration. Of course, this means that no subsequent changes to the config will be listened to at runtime. This PR fixes that issue.

fixes #2905